ElectronicCats/Beelan-LoRaWAN #4: ttn support?

halukmy, 2019-01-28:
ttn support is available?

wero1414, 2019-01-28:
Its should work with every network, but as far as i know TTN works in the subband 1 so you need to #define it instead of the subband 6

halukmy, 2019-01-30:
@wero1414 if you right, you can change the whole world brother! because this repo looks very nice,

so if you want can we work together to make it run on ttn with class-C because, i get crazy when i see class-c its so important for me...

can you help me to an example class-c for ttn?

halukmy, 2019-01-31:
@wero1414 what's your using network? https://www.loraserver.io or another?

wero1414, 2019-01-31:
as far as i know ttn does not support class C but yeah it could work, im working on refactor the way the library works, in order to implement interrupts and FHSS and some reginal parameter that this library does not have, until then i could make the class C work correctly

wero1414, 2019-01-31:
Closing this issue cause its solved, it is compatible with TTN

closed 2019-01-31 by wero1414

halukmy, 2019-02-07:
@wero1414 whats your arduino gateway code? can you share it too please? (exactly class C)

wero1414, 2019-02-08:
I have no code for and arduino gateway as you metion, i use complete lora gateways

halukmy, 2019-02-08:
What's that complete gateway ?