ElectronicCats/Beelan-LoRaWAN #8: ask

schwiderke92, 2019-02-12:
code is working?
 Where do I set the radio pinning?

wero1414, 2019-02-13:
The code works, but the functionalities that you can use can depend on the region that you configure youre library (On config.h), you can read the functionalities that depend on the region on the README.

About the radio pinning i dont catch you, what do you mean with "pinning"? 
Do you refer to the time between uplinks?

schwiderke92, 2019-02-13:
the DIO0 DIO1 pinout setting etc ...

Congratulations on the work, I'll test on my radio, I want to oscillate between class a / uplink and class c for downlink

wero1414, 2019-02-14:
Aaa ok! the pins definitions are on the Readme, you can define them 

sRFM_pins RFM_pins = {
  	.CS = SS,
  	.RST = RFM_RST,
  	.DIO0 = RFM_DIO0,
  	.DIO1 = RFM_DIO1,
  	.DIO2 = RFM_DIO2,
  	.DIO5 = RFM_DIO5
}; 

If you dont have DIO5 just define it, it does not do nothing yet

mdede439, 2019-03-31:
Hi.  I have a question about this part of the code myself.  I am new to this so maybe it is obvious to some but not to me right now.  I am trying to get this code working on a WEMOS D1 Mini and currently when I compile your code it fails with 'RFM_RST' was not declared in this scope. My understanding is SS, RFM_RST,RFM_DIO0, etc, will need to be replaced by something denoting the pins I use on WEMOS D1 Mini for my RF95 module.  If I am correct then what value do I use for the variable? For instance I want to use GPIO 15 of the WEMOS for DIO0, how do I declare that in this set of variables? Do I code ".DIO0 = GPIO15"??

sabas1080, 2019-03-31:
Simple enter pin value, example:
sRFM_pins RFM_pins = {
.CS = 10,
.RST = 11,
.DIO0 = 30,
.DIO1 = 15,
.DIO2 = 4,
.DIO5 = 33
};

mdede439, 2019-03-31:
Great, thank you.  Appreciate it.

mdede439, 2019-03-31:
And if I do not need the to use the pin do I just pick any number not corresponding to an actual pin?

wero1414, 2019-04-01:
yes

closed 2019-05-23 by wero1414