ElectronicCats/Beelan-LoRaWAN #11: MKRWAN 1300

lodesmets, 2019-02-28:
Hello,
This library looks great (and better documented than the MKRWAN library). But do you know what the pin settings are for the MKRWAN 1300 board?

sabas1080, 2019-02-28:
Hi @lodesmets 

theoretically it should work, the pins for SPI are

SPI
https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrwan1300/variant.h#L111

but you must mode the module in SPI mode as they do in the LoRa library
https://github.com/sandeepmistry/arduino-LoRa/blob/dff276e26845727049a48a624d0eb2b545f3bb6a/src/LoRa.cpp#L75

with pins
Lora RESET and BOOT

https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrwan1300/variant.h#L71

one drawback of the MKR1300 is that it does not have access to the DIO pines

lodesmets, 2019-03-01:
Thanks for the response @sabas1080, 
But the arduino lora library only requires DIO0 and this library requires DIO0-5.
What about those, what should those values be?

closed 2019-03-13 by lodesmets

wero1414, 2019-03-24:
Those DIOx pins on the MKR1300 are not connected to the samd21