ElectronicCats/Beelan-LoRaWAN #20: Testing ESP32, questions

ivancho523, 2019-08-08:
Hello, I am testing your library with an ESP32, I am loading the OTAA example and I manage to link it with my loraserver, however, the data is not always received, I am sending data every 10 seconds but sometimes minutes pass without the data arriving, additionally; When I turn the ESP32 off and on again, it can take up to 30 minutes without joining, why can that be? One last query, the channels used are from 0 to 7, however my gw is 16, how can I modify your library to access more channels? thank you

wero1414, 2019-08-16:
Hi ivan sorry for the delay answering you, first the OTAA example should init ok every reset, the thing is that if your gateway is not well configured your OTAA petition can be missing on the frequencies, also we dont have a way to activate 16 channels yet, you can add a define for the 16 channels and then change the functions instead of hardcoding the 8 channels getting the 16 channels or 8 by the definition

shmrymbd, 2020-02-18:
what is the lorawan specification version? 1.0.2?

wero1414, 2020-02-22:
yes 1.0.2