ElectronicCats/Beelan-LoRaWAN #25: Class C examples

LasaleFamine, 2020-02-21:
Hi and thanks for the library, I really like the APIs and I'm going to try it for my project.

I'm wondering, is there some specific example for working with Class C devices (EU frequency)?

wero1414, 2020-02-21:
Hi right know is kind of hard to me to make time to keep developing this library but the Class C example is the first thing on the TODO list

mr3188, 2020-02-24:
We are also testing Class C for EU_868, as TTN doesn't support class c we are preparing a ChirpStack server to test. I will update with the results.

sabas1080, 2020-02-24:
We are working on that #28 , welcome your feedback

LasaleFamine, 2020-02-24:
For Class C support I'm using The Things Stack V3, if that can help. 

Is worth noting that I'm currently not able to receive any downlink using The Things Stack V3 and this library (on Arduino Uno + Dragino Shield), also Class A downlinks are not received. Instead, I can send uplinks correctly.

For completeness, I have tested the stack with my device with LMIC library and I was able to uplink/downlink in Class A without problems.

halukmy, 2020-02-25:
@LasaleFamine  can you inform us how can we use v3 and class C on arduino and esp32

halukmy, 2020-02-25:
@mr3188  any update about chipstrack?

LasaleFamine, 2020-02-25:
@halukmy are you saying I can't receive the Class C downlink with my device? Be aware that I'm newbie here and I'm trying to learn too.

sabas1080, 2020-02-25:
@halukmy welcome your pullrequest ;)

mr3188, 2020-02-26:
With ChirpStack and class A I'm able to use join/uplink/downlink but with Class C I have problems with join using OTAA, no problems with ABP in uplinks.

halukmy, 2020-02-26:
@mr3188  can you share youtube videos how to use class c on esp32 maybe installation

mr3188, 2020-02-26:
@halukmy  I'm using a board designed by TTN Comunity of Madrid base on MattairTech  SAMD21E core.
Have you been able to use it with class A?

mr3188, 2020-02-26:
@LasaleFamine I have to check if all my changes are merged into master branch but I remember there where something  wrong with rx1 and rx2 windows delay and have in mind TTN uses a non standard DR  but I have not used ttn stack v3.

mr3188, 2020-02-27:
I can't send Class C down links ( don't even arrived to the gateway) because I have  configuration problems with my chirpStack server, it will take some time to fix it.

mr3188, 2020-03-04:
Now my server configuration is ok and I see the downlink in the gateway but not in the node.
One of the problems I have seen is class C nodes must open a small RX2 window just between the end of the uplink and the RX1 window.
What I understand from  LoRaWAN 1.03 specs for class c is:
- send uplink in the same way as CLASS A
- open small RX2 window for a second (RX1 DELAY)
- open RX1 window in the same way as CLASS A
- open RX2 window in the same way as CLASS A but leave it open (RFM in continuous receive) until next uplink.
Basically class C is the same as class A but must be listening RX2 except when transmiting and listening RX1

sabas1080, 2020-03-17:
there is an example of class C apparently working #32

LasaleFamine, 2020-03-18:
I have tested the new example with the The Things Stack V3, but still no luck unfortunately... I can see that the stack won't send down the downlink when in ABP, instead in OTAA the stack will send the downlink but the device can't receive it (I'm using an Arduino UNO with the Dragino Shield). 

@mr3188 did you test also OTAA connection with this lib and class c device, right? 

Chirpstark seems too much complex for me, I can't even correctly configure it, so I can't try actually with it.

mr3188, 2020-03-18:
Yes, I made tests with ABP and OTAA using ChirpStack and it worked as expected.
I'm not sure but I think ABP need at least one uplink before sending downlinks.
Which DR  and frequency values do you seen in gateway log for the downlink?

I haven't tested class C with TTN V3, V2 doesn't support Class C but I know TTN uses non standard frequencies for RX2. RX2 config is the default for Class C so first take a look at the data sent to the rfm module just before entering continuous receive

look for the lines shown in debug just before the following ones to check wich SF and BW is configured.

RFM_Switch_Mode 101
SPI Write ADDR: 1 DATA: 85    --> Switch mode to LoRa and continuous receive 

I will take a look at Stack V3 but I will need some time.

.

halukmy, 2020-04-02:
@mr3188 can you help me how to downlink message via chirpstack to node?

mattjbailey, 2021-12-21:
It's been some time since the post above and I haven't seen anything further... Any progress?? I also have the same issue. I'm using Chirpstack, I've used other repositories and have got my devices working... I was looking for class C support so I tried using this one. Again, uploads work fine, but downlinks do not. I have tried both class C and class A - no downlink messages are received. I can see the gateway is sending. I've also tried many different frequencies with no luck.  I am using an Arduino Mega and a Dragino shield.

wero1414, 2021-12-21:
The example is already working in Class C i personally have tested it on US915 on chirpstack and also on TTN V3 also @mr3188 have tested it

mattjbailey, 2021-12-21:
Can you post a direct link to what you tested? I would like to test it. 
This is what I see:
[image]

wero1414, 2021-12-21:
Shure this [example](https://github.com/BeelanMX/Beelan-LoRaWAN/blob/master/examples/class-C-ABP/class_c_ABP/class_c_ABP.ino) 

Seeing your screnshoot you are already receiving data cause in the main code is printed those lines, you're not seeing the data cause the way is received and the data type but the library is telling you somenthing was received.

`
if(recvStatus) {
    Serial.print("====>> ");
    Serial.println(outStr);
  }
`

mattjbailey, 2021-12-21:
I see the arrows but I get those even if not data is being sent... I'll look at your link.

wero1414, 2021-12-21:
Well you should see the data type your sending cause if your sending somenthing thats not a char the serial print wont print it

mattjbailey, 2021-12-21:
Im sending this Base64 within Chirpstack on port 2: 
SGVsbG8gV29ybGQ=
That is the string "Hello World"

mattjbailey, 2021-12-21:
Are the arrows results of it sending back a downlink message with each uplink? I'm not sure why I am seeing this? Further, the arrows show up randomly. 
[image]

wero1414, 2021-12-21:
Are you using the last version of the library?

mattjbailey, 2021-12-21:
Yes, in fact I've even checked the code changes to see they were merged in to what I'm using.

mattjbailey, 2021-12-21:
I may have figured it out... 

I had to change the loramac.cpp 

`
		// Class C open RX2 immediately after sending data
		if(LoRa_Settings->Mote_Class == CLASS_C){
			#ifdef US_915
			LoRa_Settings->Channel_Rx = 0x08;    // set Rx2 channel 923.3 MHZ
			LoRa_Settings->Datarate_Rx = SF10BW125;   //set RX2 datarate 12 // I CHANGED THIS TO MATCH WHAT WAS BEING SENT
			#elif defined(EU_868)
			LoRa_Settings->Channel_Rx = CHRX2;    // set Rx2 channel 923.3 MHZ 
			LoRa_Settings->Datarate_Rx = SF10BW125;   //set RX2 datarate 12
			#endif
			LORA_Receive_Data(Data_Rx, Session_Data, OTAA_Data, Message_Rx, LoRa_Settings);  //BUG DETECT SENDED PACKET ALWAYS (IT DOES UPDATE)`

mattjbailey, 2021-12-21:
Hopefully this helps someone. It turns out I just had to set the Rx2 datarate to 8 and things started to work. I am using Class C ABP. Thank you for your help!
[image]

mattjbailey, 2022-02-20:
The issue has not been resolved. We put this aside as we felt your product
was taking so much of our time just to make work. It’s on hold.

closed 2022-05-03 by stale[bot]