ElectronicCats/Beelan-LoRaWAN #30: How long is RFM_pins.DIO0 active upon receved?

mr3188, 2020-02-26:
With out using interrupts the Class C node knows if a downlink is received checking if RFM_pins.DIO0 i high. ¿how long does this pin remain active? I haven't found clear explanation in rfm95 datasheet.

mr3188, 2020-03-03:
Using a logic analyser it seems DIO0 is up during 87 microseconds it will be difficult to catch the pulse pulling instead of using an interrupt

wero1414, 2020-03-04:
Yes thats a problem thats why i have a branch using interupts but right now it has some mistakes

mr3188, 2020-03-04:
The pulse I detected was due to an uplink. My node is not receiving then downlink.
I will close this question until I'm sure the problem is catching the RxDone pulse in DIO0.

closed 2020-03-04 by mr3188

mr3188, 2020-03-04:
After some research I think DIO0 remains up until the MCU clears it just before reading received data.