ElectronicCats/Beelan-LoRaWAN #35: class c downlink not work with chipstark and esP-1ch-Gateway

halukmy, 2020-04-01:
greetings,
i just test chirpstack class c , uplink works from beelan, but how can  i send downlink message to chip?

and i cant continusly uplink, delay like 1 min or 40 seconds

lambertoo, 2020-04-13:
With chirpstacks application server you can send send the unconfirmed downlink as long as in the device profile you enabled that your device works in class c

mr3188, 2020-04-14:
which frequency are you using?

halukmy, 2020-04-15:
@lambertoo and @mr3188 i use eu 868, i see uplink from nodes, but can not downlink esp-1ch (node32 gateway) to nodes..

is there easiest way mqtt ? possible? can someone share screen shots how downlink

thanks

sheggy012, 2020-04-15:
@halukmy In your other issue you talk about this gateway: https://github.com/things4u/ESP-1ch-Gateway/. But the readme there says: "Selecting Class mode of operation: ... NOTE: Only class A is supported" If you find a solution, let me know. Just ran into the same problem. Here's what I've been trying to do as well: https://github.com/CongducPham/LowCostLoRaGw . Raw Lora Downlinks are working without lorawan/chirpstack. But with chirpstack it does not work for me either.

halukmy, 2020-04-18:
how downlink ?

wero1414, 2020-07-01:
There was an error on SF selection when Rx happens cause some bad macros definition we did fix it in latest commit now we're able to get Rx again

closed 2020-07-01 by wero1414