ElectronicCats/Beelan-LoRaWAN #36: Enabling ADR (  LoRaWAN Adaptive Data Rate)

lambertoo, 2020-04-13:
Hello all, i would like to thank the author of this library it’s supper easy to use without any complexity, so now i have two questions one is instead of setting a specific DataRate like lora.setDataRate(SF8BW125) is there a way we can let a node decided which DataRate is best for it according to the distance between node and Gateway.

My second question is i was checking in the library function but i haven’t seen any way we can put a node into sleep mode without putting the whole microcontroller into sleep mode. Thanks

mr3188, 2020-08-19:
Hi lambertoo, ADR is set by server and sent to the node using MAC commands  I have plans to implement it in this library if the authors aren't already doing it but first need to add MAC commands functionality. I'm still checking differences in regional parameters implying changes in functionality from EU868 (my region) and US915  (original library region)

daeynasvistas, 2020-11-13:
@mr3188 .. do you have some news on ADR enhancement? 
It's a very useful feature ;)

mr3188, 2020-11-13:
I'm sorry  I have been busy looking at confirmed downlink/uplink issues but I will try to work on MAC commands common code to be able to have ADR on top.

With ADR the server ask the node to change DR but there are some SX12... registers that give us information of last messages signal strength. i haven't used it before  but I suppose it can help the to the node decide which DR to use.

I will first try with LinkCheck MAC command that seems to be simpler and then go forward with ADR.

mr3188, 2020-11-20:
I have successfully sent LinkCheckReq command from the node to the ChirpStack server as a single uplink with port=0 and 0x02 in payload. ChirpStack recognises the command but I can't see any downlink with LinkReqAns from the server.

I don't know if the problems is the ChirpStack server or the gateway (RAK831), and I will not have another gateway to test until next week.
With TTN v2 I don't see any information about the command, just an uplink without payload.

I will try sending the command together with a regular uplink using Frame options field.
Does any body know how to trace communication between ChirpStack server and the gateway?

mr3188, 2020-11-23:
I have changed the gateway to a Dragino LPS8 and know I can see LickCheckAns dow links. I will confirm I'm handling LinkCheckAns correctly before pushing changes to repo.

mr3188, 2021-02-12:
I have created #75 as a base to implement ADR support.
It has some MAC commands already working and tested and I will commit changes to be able to discuss if it meets al regional requirements we need to change frequencies, active channels and data rates now stored in static structures.

closed 2021-07-05 by stale[bot]