ElectronicCats/Beelan-LoRaWAN #37: Support AU_915

rogerlucas, 2020-04-21:
Hello, I am working on a project with the frequency AU_915. I would like to contribute the necessary changes for it, but I confess that I am a little lost and did not understand the relationship between the values of the array with the frequencies of each channel.

```
static const PROGMEM unsigned char LoRa_TX_Freq [8] [3] = {
{0xE2, 0x60, 0x26}, // Channel [0], 905.5 MHz / 61,035 Hz = 14835750 = 0xE26026
```

Can you please help me on this most basic point?

sabas1080, 2020-04-21:
Hi @rogerlucas 

For this feature it is necessary to add the band AU915, not only datarate, but I think for AU915 it is

```
#if defined(AS_915)
    SF12BW125   = 0x00,
    SF11BW125    = 0x01,
    SF10BW125    = 0x02,
    SF9BW125    = 0x03,
    SF8BW125    = 0x04,
    SF7BW125   = 0x05,
    SF8BW500   = 0x06,
    SF12BW500   = 0x08,
    SF11BW500    = 0x09,
    SF10BW500    = 0x0A
    SF9BW500    = 0x0B
    SF8BW500    = 0x0C
    SF7BW500    = 0x0D
```

if you need help to add the band AU915 please let us know, we look forward to your pullrequest

rogerlucas, 2020-04-21:
> Hi @rogerlucas
>
> For this feature it is necessary to add the band AU915, not only datarate, but I think for AU915 it is
>
> ```
> #if defined(AS_915)
>     SF12BW125   = 0x00,
>     SF11BW125    = 0x01, [...]

Thank you very much. When I have news I will return here.

wero1414, 2020-07-01:
Hi @rogerlucas in the last commit we add the AU915 i would like that you test this recent upgrade

closed 2020-07-01 by wero1414