ElectronicCats/Beelan-LoRaWAN #38: Pregunta??

RCB07, 2020-04-30:
Con que tarjetas de desarrollo se pude utilizar?... Actualmente cuanto con una nodemcu, una stm32 y un arduino pro mini

Espero su respuesta, Saludos
Excelente Aportación

sabas1080, 2020-05-07:
Hola @RCB07 perdón por la tardia respuesta (me sigo tratando de acostumbrar a las nuevas notificaciones de github), en teoria debería soportar estas arquitecturas  nuestros test automatizados han pasado la pruebas https://github.com/BeelanMX/Beelan-LoRaWAN/actions/runs/77583123

RCB07, 2020-05-07:
Ya hice funcionar la libreria en un arduino pro mini, ahora tengo problemas al recibir los datos.
Usando el ejemplo de abp de la misma libreria deberia mandar datos cada 10 segundos pero tarda de 4 a 5 minutos y en lapsos incluso mas.
He probado todas las sub bandas será problema de la libreria o del gateway..

[image]

Espero y me puedan ayudar, solo estoy cambiando las sub bandas y usado el sf7 bw150

RCB07, 2020-05-07:
Saludos y muchas gracias..

sabas1080, 2020-05-07:
Que version de arduino pro mini usas 3.3v o 5v? Tu gateway es multicanal o mono canal?

RCB07, 2020-05-07:
Uso el 3.3v y estoy usando un gateway monocanal basado en una raspberry pi

sabas1080, 2020-05-07:
@RCB07 hay dos posibles cosas que esten pasando
1.- Debes transmitir solo en el canal de tu gateway monocanal, la biblioteca por default transmite en 8 canales de manera aleatoria mientras hace el recorrido de todos los canales puede tardar varios minutos.
2.- la otra el reloj del Pro Mini a 3.3v puede que este cambiando los tiempos de envio en los que se basa la biblioteca y se tarda mas de 10 segundos.

Apuesto mas por la primera, creo debes estar transmitiendo en 8 canales

RCB07, 2020-05-07:
Seria cambiar esta linea de codigo entonces..
[image]
y 
 
Agradezco muchísimo tu ayuda en una tecnología tan nueva y tan poco explorada en México y con tan poca información en el tema.... Soy estudiante de ing.electrónica con la disposicion de aprender cosas nuevas :)

RCB07, 2020-05-07:
Funcionó perfecto, Gracias de nuevo

closed 2020-05-07 by RCB07

sabas1080, 2020-05-07:
De nada