ElectronicCats/Beelan-LoRaWAN #40: ACK reception

dberenguer, 2020-06-11:
What is the way to check if an uplink has been acknowledged?

Thanks!!

mr3188, 2020-06-12:
Hi Daniel,
Let me check but as I remember the uplink ack is just an  empty downlink.
I
Miguel Romaní

dberenguer, 2020-06-12:
That's probably right but readData makes no difference between no packet received and packet received with payload length = 0.

mr3188, 2020-06-12:
You are wright, LoraMAC sets Buffer_rx.Counter to the payload length, not the packet length so readData thinks there is no message.
Control frame ACK flag is not being checked so there is no way to see if the last message has been confirmed by server.

But I remember, when testing class c nodes to see empty messages  in output trace. I will test with this node again, may be it was only debug traces from other parts of the code.

Anyway, I will review LoraWAN specs to find out the correct workflow

Miguel

dberenguer, 2020-06-12:
Thanks Miguel,

Checking ACK from update() instead of readData() and then passing the ACK status to readData through a class attribute should be straightforward. If you agree I can work on a pull request.

mr3188, 2020-06-12:
It would be great

dberenguer, 2020-06-16:
Created [pull request](https://github.com/BeelanMX/Beelan-LoRaWAN/pull/41)

closed 2020-06-17 by wero1414