ElectronicCats/Beelan-LoRaWAN #42: Encode/Decode

simortus, 2020-07-26:
Hi, thanks for your contribution,
Can you tell what encoding are you using? i get unreadable chars on chirpstack! i want to add a chirpstack  repo when I finalize this.
If you are around in Copenhagen, lets get that beer 👍

sabas1080, 2020-07-27:
Hi @simortus What example are you using? Beautiful Copenhagen, I hope to return soon

sabas1080, 2020-08-16:
closed due to inactivity

simortus, 2020-08-19:
Hey,
Sorry about the late reply, got busy with work and holidays. I hope you had
good time as well:)
I am using the abp example. I am using chirpstack lorawan server. Typically
base64 is the go for, but no results.
I use this to decode.
let me know if I need to change something?
Cheers.
var result = ""; for(var i = 0; i < bytes.length; ++i){ result+=
(String.fromCharCode(bytes[i])); } var decodedStr =
(result).toString('base64'); // var decodedStr = (result).toString();
return decodedStr;

sabas1080, 2020-08-19:
@wolftrax5 can you help here?

simortus, 2020-08-19:
Where ?

wolftrax5, 2020-08-19:
how you are implementing that function?  and witch version of her chirpstack server are you using, can you share how are you implementing. the bytes should be an array of numbers, if you receive strings this will never return somethings and on a resents versions of chirpstack server, has a template of decode function.

`const data = [65,66,67]

function decode (bytes) {
  var result = ''
  for (var i = 0; i < bytes.length; ++i) {
    result += (String.fromCharCode(bytes[i]))
  }
  var decodedStr = (result).toString('base64')
  // var decodedStr = (result).toString();
  return decodedStr
}
console.log(decode(data))
//ABC
`
[another way to decode](https://gist.github.com/wolftrax5/d6a20b893262e3a60c010d3f4f39ab79)

closed 2020-10-14 by sabas1080