ElectronicCats/Beelan-LoRaWAN #43: TTN Port Number

barni007-pro, 2020-08-14:
Thank you for the great library.
I would like to be able to specify the port for the next version.
Is there any way I can do this now.

barni007-pro, 2020-08-15:
My solution:

lorawan-arduino-rfm.h
`void sendUplink(char *data, unsigned int len, unsigned char confirm, unsigned char mport = 1);`

lorawan-arduino-rfm.cpp
```
void LoRaWANClass::sendUplink(char *data, unsigned int len, unsigned char confirm, unsigned char mport)
{
    if (currentChannel == MULTI) {
        randomChannel();
    }
    LoRa_Settings.Confirm = (confirm == 0) ? 0 : 1;	
    if (mport == 0) mport = 1;
    if (mport > 223) mport = 1;	
    LoRa_Settings.Mport = mport;
    //Set new command for RFM
    RFM_Command_Status = NEW_RFM_COMMAND;   
    Buffer_Tx.Counter = len;
    memcpy(Buffer_Tx.Data,data,len);
}
```

Struct.h
```
typedef struct {
    unsigned char Confirm;		//0x00 Unconfirmed, 0x01 Confirmed
    unsigned char Mport;                //Port 1-223
    unsigned char Mote_Class;		//0x00 Class A, 0x01 Class C
    unsigned char Datarate_Tx;		//See RFM file
    unsigned char Datarate_Rx;		//See RFM file
    unsigned char Channel_Tx;		//See RFM file
    unsigned char Channel_Rx;		//See RFM filed
    unsigned char Channel_Hopping;	//0x00 No hopping, 0x01 Hopping
    unsigned char Transmit_Power;	//0x00 to 0x0F
} sSettings;
```

LoRaMAC.cpp
```
void LORA_Send_Data(sBuffer *Data_Tx, sLoRa_Session *Session_Data, sSettings *LoRa_Settings)
.......
//RFM_Data[8] = Message.Frame_Port;
RFM_Data[8] = LoRa_Settings->Mport;
.......

```
Can you build that into the next version?
Thank you and keep it up.

wero1414, 2020-08-16:
Hey @barni007-pro looks great, did you already test it? i dont have any node near me until monday so i will add as soon as i test it out! thanks for the issue and the solution!

sabas1080, 2020-08-16:
I have added pullrequest for testing https://github.com/BeelanMX/Beelan-LoRaWAN/pull/44, thanks @barni007-pro

barni007-pro, 2020-08-16:
Everything is going fine in my project.
My location is Germany and everything arrives at "The Things Network TTN".

sabas1080, 2020-08-18:
solution in #44 , thanks @barni007-pro

closed 2020-08-18 by sabas1080