ElectronicCats/Beelan-LoRaWAN #46: Confirmed downlink for Class-C devices

bkaraceylan, 2020-09-07:
Hello,

Using chirpstack, I am unable to get acknowledgement for confirmed downlink requests for Class-C devices. Is this not supported or am I doing something wrong?

Thanks,

iotpanama, 2020-10-23:
Los enlaces descendientes no funcionan... Como se puede corregir esto

plachta-zabatt, 2020-10-27:
I will have to test later but they are working for Class-A devices I believe. I at least am fairly certain that they show up in device_ack table if you implement postgresql integration w/ chirpstack.

sabas1080, 2020-10-28:
Hi everyone, we have tested the solution to this problem, help us to test

iotpanama, 2020-10-28:
> Hi everyone, we have tested the solution to this problem, help us to test

How can we do the tests?

mr3188, 2020-10-28:
I'm also implementing a solution for confirmed downlinks I have tested it with Class A and will begin testing Class C  in the next days.
My main question is when to send the ACK from the node to the gateway as the LoRaWAN specification says the node can send it immediately in an empty downlink  or deferred to the next schedules uplink. As my tests with TTN shows the server is continuously re sending the downlink until the node sends and ACK I think I will set a flag so the first uplink after a confirmed downlink will implicitly include de ACK bit set.

mr3188, 2020-10-28:
One additional question. How does ttn stack v3 manage confirmed downlinks? in v2 Class A devices the downlink is sent every time until and ACK from the node is received. As Class C device remains listening most of the time, when is the downlink resent? Is it also  sent only after an uplink?

iotpanama, 2020-10-28:
> Enviado el: miércoles, 28 de octubre de 2020 3:50 a. m.

Hi Miguel

Indeed, the behavior would be that the gateway automatically sends an ACK automatically in the time window opened by the device. This is configured in the sending function, when you expressly request a confirmation from the gateway. This is very useful in class A devices as it allows a just moment for the sensor to go into deep sleep.

 

____________________________

 

Ing. Jorge Quevedo

Movil +50763280221

Movil +50762028069

Web:  <http://www.iotpanama.com/> www.iotpanama.com

Panama city - Panama

 

Por favor, tenga en cuenta su responsabilidad ambiental.

Antes de imprimir este mensaje de correo electrónico, pregúntese si realmente necesita una copia en papel.

 

El contenido de este mensaje es confidencial.

Este mensaje no tiene carácter vinculante.

Este mensaje va dirigido, de manera exclusiva, a su destinatario y contiene información confidencial sujeta al secreto profesional cuya divulgación no está permitida por la ley. En caso de haber recibido este mensaje por error le rogamos, que de forma inmediata, nos lo comunique mediante correo electrónico remitido a nuestra atención y proceda a su eliminación así como a la de cualquier documento adjunto al mismo.

wero1414, 2020-10-30:
@mr3188 @iotpanama @bkaraceylan i started the PR #58 fixing the timming problems for the downlinks, by far i got the Class C working well, and i think Class A but i need to do more testing, the PR is now just for US but feel free to start adding your regions, so we all can test! thanks!

daeynasvistas, 2021-06-29:
.. is this working? never was able to confirm ack using chirpstack
i can see the ack in the log gateway, but readAck() never seams to catch the value.

dhtek, 2021-07-16:
> 
> 
> .. is this working? never was able to confirm ack using chirpstack
> i can see the ack in the log gateway, but readAck() never seams to catch the value.

Same for me. I'm using with EU868, CLASS A with TTNB V3 and lora.readAck() return always false.

`Serial.print("Sending: ");
    Serial.println(myStr);
    
    lora.sendUplink(myStr, strlen(myStr), 1, 1);

    counter++;
  }

  ackStatus = lora.readAck();
  if (ackStatus) {
    Serial.println("Acknowledged");
  }

  recvStatus = lora.readData(outStr);
  if (recvStatus)
  {
    Serial.println(outStr);
  }
 
  // Check Lora RX
  lora.update();`

sabas1080, 2021-07-16:
@daeynasvistas the changes in #58  the pullrest are in the latest release