ElectronicCats/Beelan-LoRaWAN #47: stm32

tcpipchip, 2020-09-11:
plans to STM32 L0 + SX1276 ?

If yes, i would like to test here!

tcpipchip, 2020-09-11:
I see that compiles!!!

Where is in your code make this config ?

#if defined (STM32L071xx)

#define RADIO_RESET                         PA_8

#define RADIO_MOSI                          PA_7
#define RADIO_MISO                          PA_6
#define RADIO_SCLK                          PA_5
#define RADIO_NSS                            PA_4      

#define RADIO_DIO_0                         PA_12		
#define RADIO_DIO_1                         PB_2		
#define RADIO_DIO_2                         PB_5		
#define RADIO_DIO_3                         PB_0		
#define RADIO_DIO_4                         PA_11		
#if defined (USE_TCXO)
#define ENABLE_TCXO							PB_1
#else
#define RADIO_DIO_5                         PB_1		
#endif	

#define RADIO_ANT_SWITCH_HF                 PA_15

sabas1080, 2020-09-14:
Hi @tcpipchip 
Check API in the README

```
sRFM_pins RFM_pins = {
  	.CS = SS,
  	.RST = RFM_RST,
  	.DIO0 = RFM_DIO0,
  	.DIO1 = RFM_DIO1,
  	.DIO2 = RFM_DIO2,
  	.DIO5 = RFM_DIO5
}; 
```
For config https://github.com/BeelanMX/Beelan-LoRaWAN/blob/master/src/arduino-rfm/Config.h

tcpipchip, 2020-09-15:
Thanks! I hope start the tests next week

tcpipchip, 2020-09-16:
Hi 
Got the SX1276 today!
I see that i have
https://www.nicerf.com/Upload/ueditor/files/2020-03-05/LORA1276-100mW%20long%20range%20Spread%20Spectrum%20modulation%20wireless%20transceiver%20module%20V2.2-bdc587bf-32c2-413c-8e4f-a309a8a00abd.pdf

DIO0,DIO1,DIO2 ONLY, NOT DIO5

is that a problem to use with you LIB ?

tcpipchip, 2020-09-17:
Good news!!!! U-BLOX NINA B302 (NRF52840) is acessing the LoRaWAN :)
[image]
[image]

tcpipchip, 2020-09-17:
Good news!!!! Success using the NINA B302 to access the LoRaWAN :)
[image]
[image]

tcpipchip, 2020-09-17:
[image]

tcpipchip, 2020-09-17:
NINA B302 is NRF52840 core!

tcpipchip, 2020-09-17:
[image]

sabas1080, 2020-09-18:
Nice thanks @tcpipchip !

closed 2020-09-18 by sabas1080