ElectronicCats/Beelan-LoRaWAN #50: samR34 xplained pro + IN865 + OTAA + CLASS A

supersunil, 2020-10-08:
Hi,
I am currently working on samR34 xplained pro LoRa shield .  I need to be work in IN865 with OTAA Class A. 
I have seen that you have worked on few Bands. Is it possible to have IN865 BAND update. or please guide to do 

What are all the changes need to be taken to implement this IN865 band to work with OTAA CLASS A concept

sabas1080, 2020-10-08:
Hi @supersunil 

We have not worked in IN865, your contributions to implement it are welcome

supersunil, 2020-10-09:
@sabas1080 - Thank you for your quick response. Can you guide me the process how you've done it for EU868 which is very similar to IND865, that would be really helpful for me.

mr3188, 2020-10-09:
Hi @supersunil,
First step is to download regional parameters document from LoRa Alliance web:
https://lora-alliance.org/resource-hub/rp2-101-lorawanr-regional-parameters-0

For EU868 I used an older version of the document and I have just realised there is a new (I will have to review).

Edit RFM95.cpp and ad a block for IN865 frequencies, you can copy EU868 and change values according to you freq plan.

Review also datarates and channels values, the behaviour of the code seems to be the same in both regions.
I also reviewed TTN regional plans (if you use it) because in some cases it doesn't follow LoRa Alliance standard.

supersunil, 2020-10-11:
@mr3188 - Thank you for your guidance. I've managed to configure the region to IN865 and successfully joint to the server and sent the payload.

When I hit the server with the join request and transfer the data, is there a way to get the acknowledgement from the server when the data been received? Kindly help me on that.

PS: I've tried both Confirmed and Unconfirmed method to send the data to the server, but haven't received any acknowledgement.

supersunil, 2020-10-11:
My second question is about the size of the payload. When my size of the payload is 54 characters, the MIC encryption handles it and the data been sent to the server successfully, what I need to do to send data more than 54 characters, let's say I need to send over 266 characters.

Kindly help me on that also.

mr3188, 2020-10-11:
Maximun datasize is given in he regional parameters document

mr3188, 2020-10-11:
regarding the uplink ACK I have to check because It should be implemented but I haven't tested it.

sabas1080, 2020-10-11:
@supersunil if you have modified the library to work in IN865, it would be great if you could submit your changes as a contribution

supersunil, 2020-10-11:
@sabas1080 - I would love to, right now I'm performing the testing, once it's done, I'll submit it.

sabas1080, 2020-10-11:
I have also created a variant for SAMR34 Xplained PRO in our Arduino core

https://github.com/ElectronicCats/ArduinoCore-samd/pull/3

supersunil, 2020-10-12:
Dear @mr3188, @sabas1080 

Kindly help with the payload size. 

When I try to hit the server with 54 (characters) bytes of data it was successful. But when I try to hit it with more than 54 (characters) bytes, the controller resets (Included the log)

On the server side, the gateway status changed to "mic_error"

Console Log:

`--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM5  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
SYSTEM_START
Ready
Joining...
Joined to network
BUTTON_INITT
TWI_INIT
AHT10 init
AHT10 OK
AHT 10 Sensor Initialization complete
RESULT : C0
Body Temprature Sensor Initialization complete
powerOn
Accelerometer Sensor Initialization complete
Place your index finger on the sensor with steady pressure.
Heart Rate Sensor Initialization
TIMER_START
SensorInitialize : 1
Sending: #ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFG,62 OK
SYSTEM_START
Ready     
Joining...
Joined to network
BUTTON_INITT
TWI_INIT    
AHT10 init
AHT10 OK
AHT 10 Sensor Initialization complete
RESULT : C0 
Body Temprature Sensor Initialization complete
powerOn
Accelerometer Sensor Initialization complete
Place your index finger on the sensor with steady pressure.
Heart Rate Sensor Initialization
TIMER_START
SensorInitialize : 1 
Sending: #ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFG,62 OK
SYSTEM_START
Ready
Joining...
Joined to network
BUTTON_INITT
TWI_INIT
AHT10 init
AHT10 OK
AHT 10 Sensor Initialization complete
RESULT : C0
Body Temprature Sensor Initialization complete
powerOn
Accelerometer Sensor Initialization complete
Place your index finger on the sensor with steady pressure.
Heart Rate Sensor Initialization
TIMER_START
SensorInitialize : 1 
Sending: #ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFG,62 OK
SYSTEM_START
Ready
Joining...
Joined to network
BUTTON_INITT
TWI_INIT
AHT10 init
AHT10 OK
AHT 10 Sensor Initialization complete
RESULT : C0
Body Temprature Sensor Initialization complete
powerOn
Accelerometer Sensor Initialization complete
Place your index finger on the sensor with steady pressure.
Heart Rate Sensor Initialization
TIMER_START
SensorInitialize : 1 
Sending: #ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFGHIJKLMNOPQRSTUVWXYZ#ABCDEFG,62 OK`

mr3188, 2020-10-13:
Max payload size is fixed now to 64 bytes but not checked when sending messages. As a work around you can change this value in  src\arduino-rfm\lorawan-arduino-rfm.h line 95 and 97 but I haven't tested it yet.

supersunil, 2020-10-14:
@mr3188 -  Thanks for your response man. I'll check it and revert back to you.

supersunil, 2020-10-26:
I can't able to read the acknowledgement bit from the server and i can't able to receive any data from the server. is there any configuration need to done on server side or controller side?

plachta-zabatt, 2020-10-26:
I got the payload to successfully inc to 230 by doing the changes below. I have been running onthis without issue for a a couple weeks sending many packages.

in encrypt.cpp line 173 increase size of MIC_Data to 268 (might not be perfect but this was my math)

in LoRaMAC.cpp change line 124 increase size of RFM_Data to 230 from 64 -- line 261 change RFM_Data to 230 from 64 

in lorawan-arduino-rfm.h line 95 increase size of Data_Tx to 230 from 64 -- line 97 change the size of Data_Rx to 220 from 64

This will give you 230 up and 220 down.

plachta-zabatt, 2020-10-26:
Also, in LoraMAC.cpp, on line 442 we added randomSeed(micros()); before taking a random number there to avoid dev nonce errors on join. It was getting impossible to test things w/o doing this (We were joining every time for testing purposes)

supersunil, 2020-10-26:
I will check out 

Have you tried to read acknowledge bit from server when you send the data?.
Have you receive any other data from the server?

plachta-zabatt, 2020-10-26:
We haven't messed too thoroughly with the ack stuff-- but haven't had any issues using it in passing. Same with sending data to the node from the server. We had issues initially because we were trying to send pieces of data that were too big to the node and it would crash the program. The size fixes above stopped that from happening. If you're having trouble with downlinks maybe try sending a very small bit of data to the node and see if that works. If it does, it's just the size issue crashing your program.

Ajaynithish, 2020-10-31:
Hi @plachta-zabatt I too testing this code for past two days and faced the same issue for payload. Made the changes to the files in library you have mentioned ,now my entire uplink is not working even with 50bytes. what is the controller and lora modem you were using?

sabas1080, 2020-11-01:
@Ajaynithish the lasted version has patch for max payload, download from github https://github.com/BeelanMX/Beelan-LoRaWAN/pull/57

Ajaynithish, 2020-11-02:
Hi @sabas1080. I have tried the new update. There may be a issue in the way the message format is created or they are not emptying the buffer for next message. Because device address is keep on changing for every message expect first one. And even for first message I am  getting UPLINK MIC(invalid MIC) error. Did you faced any issues like this.
[image]

closed 2021-07-05 by stale[bot]