ElectronicCats/Beelan-LoRaWAN #51: arduino SX1276

ebelouet, 2020-10-09:
HI,
Where can i find setup for SX1276 chip example because we have only for RFM95W chip ?

Thank
Eric

mr3188, 2020-10-09:
Registers in RFM95 and SX1276 datasheets seems to be the same it should work. In fact, RAK4260  module (I used to test) has SX1276 , not a hole FRM95 module. 
You will just need to check pinout.

ebelouet, 2020-10-09:
thank, i chekc it now , and i see that, i will try it,

Koraze, 2020-10-09:
it seems it kind of works with the "send-class-A-OTAA" example and my equipment : 

- Chip / end node LoRa : Heltec Lora32 chip (ESP32 + SX1276) 
- Contry : EUROPE 868MHz (I changed config.h to set `#define EU_868`)
- LoRaWAN gateway : (Local) Indoor Dragino LSP8
- LoRaWAN server : (Local) ChripStack environment on a Raspberry Pi (Ubuntu Server) ...

I said "kind of" because :

- it connect fast the first time (join) but takes times (about 30min) after a reboot
- half of the message the lora node send is lost :(

I changed this in the example code :

```
const sRFM_pins RFM_pins = {
  .CS = 18, //j
  .RST = 14, //j
  .DIO0 = 26, //j
  .DIO1 = 35, //j
  .DIO2 = 34, //j
  .DIO5 = 40, //?
};
...
  SPI.begin(5,19,27,18);
  if(!lora.init()){
    Serial.println("RFM95 not detected");
    delay(5000);
    return;
  }
```

ebelouet, 2020-10-09:
Hi
Thank a lot, I will try it Monday.goood week end.Eric

ebelouet, 2020-10-09:
so, I need to connect spi bus and io pins, do you have schematic example for this ?

sabas1080, 2020-10-10:
You need pins SPI port and pins for SX1276 or RFM95

ebelouet, 2020-10-10:
my question is to know if this pin connections is good:

ESP32 Pins | SX1278 Pins
-- | --
GND | GND
3.3V | VCC
D5 | NSS
D23 | MOSI
D19 | MISO
D18 | SCK
D14 | RST
D2 | DIO0

Koraze, 2020-10-10:
sorry I dont have schematic example for this, my Heltec Lora32 v2 is a chip with both ESP32 and LoRa together, see this link : 

- https://community.hiveeyes.org/uploads/default/original/2X/e/eda042713108809e3511e822a1aa4e582ee70ebc.png

What I can say is

- The code seems to work even without this line `SPI.begin(5,19,27,18);`
- I dont have this pin in my chip, so I write a non pin value `.DIO5 = 40,`
- I think I lost half of the messages because of my LoRa gateway (it seems to reboot every 30s) instead of this library

ebelouet, 2020-10-10:
thank, i will try it monday and give you answer.Eric

 
 

   
   - https://community.hiveeyes.org/uploads/default/original/2X/e/eda042713108809e3511e822a1aa4e582ee70ebc.png

ebelouet, 2020-10-12:
HI,Please add SPI.begin(5,19,27,18); in your example to run very well.Now, I have connection on thingsnetwork, but I have only connection but i m receive nothing on web site.Eric

 

 
 

   
   - https://community.hiveeyes.org/uploads/default/original/2X/e/eda042713108809e3511e822a1aa4e582ee70ebc.png

ebelouet, 2020-10-16:
HI
I don't know why no I have only connection on TTN and no data are sending with your demo send  a otaa ?
Do you know where is problem?Thank Eric
    Le lundi 12 octobre 2020 à 11:19:44 UTC+2, eric belouet <ebelouet@yahoo.fr> a écrit :  
 
  HI,Please add SPI.begin(5,19,27,18); in your example to run very well.Now, I have connection on thingsnetwork, but I have only connection but i m receive nothing on web site.Eric

 

 
 

   
   - https://community.hiveeyes.org/uploads/default/original/2X/e/eda042713108809e3511e822a1aa4e582ee70ebc.png

ebelouet, 2020-10-16:
HI,
I also tested RFm95w and it's ok for connextion to TTN, but I have only 1 on that can send data, another modules only connect without sending data with same software, and 4 minutes for connections.
Do you know where is problem of it ?
Thank
eric

sabas1080, 2020-10-28:
review your chosen communication channels

closed 2020-10-30 by sabas1080

Koraze, 2020-11-17:
I found another issue, maybe it will help
My Heltec Lora32 v2 didn't succeed to join if I set this line : lora.setDataRate(SF9BW125);
But if I remplace SF by another value, for example 8 (lora.setDataRate(SF8BW125);) now it works
And it works event if after join, I set the SF value to 9 again