ElectronicCats/Beelan-LoRaWAN #54: ESP-12F (esp8266) Soft WDT reset try Join

daeynasvistas, 2020-10-24:
Well it seems that i still have problem when join with OTAA:
ESP-12F with SX1276

```
const sRFM_pins RFM_pins = {
  .CS = 15,
  .RST = -1,
  .DIO0 = 5,
  .DIO1 = 4,
  .DIO2 = -1,
  .DIO5 = -1,
};
```
Serial.print()  give:
```
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
Joining...
```

ESP reset and try to join again

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 01a0
3ffffec0:  3ffee5ec 3ffee560 3ffee604 402022c9  
3ffffed0:  3ffee5ec 3ffee560 3ffee5b4 40201d50  
3ffffee0:  3fffff3c 00000005 3ffee604 402023e3  
3ffffef0:  00000000 4bc6a7f0 eccccccc 00000000  
3fffff00:  00000000 00000000 4bc6a7f0 00000000  
3fffff10:  3ffe878e 00000000 40100200 00001fc0  
3fffff20:  3ffffee0 00000000 00000000 00000000  
3fffff30:  0024b711 654c8000 3f346068 3fffff24  
3fffff40:  00000017 3ffee668 3ffe878c 00000000  
3fffff50:  3ffee604 3ffee5ec 3ffee4d8 402028aa  
3fffff60:  00001510 3ffee560 3ffee5b4 4020308c  
3fffff70:  3ffee668 3ffee668 3ffee668 3ffe861c  
3fffff80:  3ffee668 3ffee668 3ffee4d8 402010fb  
3fffff90:  feefeffe feefeffe feefeffe 3ffee6d4  
3fffffa0:  3fffdad0 00000000 3ffee694 40203804  
3fffffb0:  feefeffe feefeffe 3ffe84ec 40100c29  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

i use example scketch
```
 *  Test in all frequencies of US_915 band
 *      Using SUB BAND 1 [ 903.9 - 905.3 Mhz] see config.h
 *      channel [0 - 7]
 *      data rate [0 - 3]
 *  Class A and C devices with OTAA activation
```

daeynasvistas, 2020-10-24:
Ok so it seems its the ESP8266 yield() "problem"

I try to put yield(); in do..while but unable to fix the problem

i need to add yield in the **LoRaMAC.cpp** file

```
bool LORA_join_Accept(sBuffer *Data_Rx,sLoRa_Session *Session_Data, sLoRa_OTAA *OTAA_Data, sLoRa_Message *Message, sSettings *LoRa_Settings)
{
(...)

	//RFM to single receive
	Message_Status = RFM_Single_Receive(LoRa_Settings);  
	//If there is a message received get the data from the RFM
	if(Message_Status == NEW_MESSAGE)
		Message_Status = RFM_Get_Package(&RFM_Package);

yield(); //<--- ESP8266 problem

	//if CRC ok breakdown package
	if(Message_Status == CRC_OK)
	{
		//Get MAC_Header
    	Message->MAC_Header = RFM_Data[0];

(...)
```

then reset goes away and joining gets normal loop .. joining ... joining ... joining

but add yield() in main library its not a good solution .. do anyone have better ideia?

sabas1080, 2020-10-24:
@daeynasvistas you can add the function for ESP8266 or ESP32 as

```
#if defined(ESP8266) || defined(ESP32) 
yield();
....
#endif
```

wero1414, 2020-10-27:
Closed due the yield function is already merged on loramac file

closed 2020-10-27 by wero1414