ElectronicCats/Beelan-LoRaWAN #59: Arduino Uno + RFM95W configured for the IN865_867 band getting stuck on 'Joining...'

SiddhantArya, 2020-11-03:
Hi, 
I have an Arduino Uno and RFM95W (868 MHz) version and I've tried to configure the code for the Indian band (865-857 MHz), but the code appears to be stuck on 'Joining...'.

I've made a few changes to the code as some had suggested on another thread. 

Firstly in config.h, I have added the support for the IN865_867 band as below:
`#define IN_865_867

#ifdef IN_865_867
    //Select the subband youre working on
    // make sure your gateway is working in one of these bands
    //#define SUBND_0    
    #define SUBND_1     
    //#define SUBND_2  
    //#define SUBND_3  
    //#define SUBND_4  
    //#define SUBND_5  
    //#define SUBND_6  
    //#define SUBND_7  
#endif
`
Then, I changed the enumeration in Struct.h to have the parameters for the IN865_867 band:
`#elif defined(IN_865_867)
    SF12BW125   = 0x00,
    SF11BW125   = 0x01,
    SF10BW125   = 0x02,
    SF9BW125    = 0x03,
    SF8BW125    = 0x04,
    SF7BW125    = 0x05
`
In the file RFM95.cpp, I have added this code to the LoRa Frequencies section, while keeping the RFM_Change_Datarate function as it is.
`#elif defined(IN_865_867)
  static const PROGMEM unsigned char LoRa_Frequency[4][3] = {//[865.1 - 866.9] MHz
    { 0xD8, 0x44, 0x24 }, //Channel [0], 865.0625 MHz / 61.035 Hz = 14173220 = 0xD84424
    { 0xD8, 0x57, 0x8F }, //Channel [1], 865.4025 MHz / 61.035 Hz = 14178191 = 0xD8578F
    { 0xD8, 0x7F, 0x2F }, //Channel [2], 865.9850 MHz / 61.035 Hz = 14188335 = 0xD87F2F
    { 0xD8, 0xA3, 0x58 }, // RX2 Receive channel 866.550 MHz / 61.035 Hz = 14197592 = 0xD8A358
  };
`

Furthermore, I have kept the RFM_Change_Channel code for IN band similar to the EU band as below
`#elif defined(IN_865_867)
  if (Channel <= 0x08)
    for(unsigned char i = 0 ; i < 3 ; ++i)
      RFM_Write(RFM_REG_FR_MSB + i, pgm_read_byte(&(LoRa_Frequency[Channel][i])));
`

Finally, I am using the pin map as shown below:
`
const sRFM_pins RFM_pins = {
   .CS = 10,
  .RST = 5,
  .DIO0 = 7,
  .DIO1 = 6,
  .DIO2 = 2,
  .DIO5 = 3,
};
`

I have a Pycom gateway configured to be listening on the Indian LoRa band. I can see the join request messages on the TTN console, however, I don't see any messages from the node after the join request and the Serial Monitor shows that the Node is stuck on 'Joining...'. 

Can someone tell me what else needs to be done to get this to work?

wero1414, 2020-11-04:
Hi! i would suggest you to first try ABP session with your new configurations, then you will also need to change the datarate configuration i think so it will be the same than your gateway

mr3188, 2020-11-04:
I think IN_865_867 is similar to EU_868, don't have subbands. It is better to begin from EU_868 config, change frequencies table according to your region and 

@wero1414 is right, try first with ABP.

In TTN console check if the join server is sending JoinAccept, if not, check AppEUI and DevEUI are the same in the node and ttn console.

SiddhantArya, 2020-11-11:
I can connect the gateway using ABP. I can see the packets being received with the correct payload too..

Here's a snippet:

[image]

manivalaguru, 2021-04-29:
hello,
i have you get downlink on your node too? i need some help on this
thanks and regards
manikandan

SiddhantArya, 2021-05-03:
No, I have still not gotten a solution. I ended up buying an Arduino Nano to check whether it is a memory issue

manivalaguru, 2021-05-03:
it is working i can able to get downlinks in my node

SiddhantArya, 2021-05-03:
Can you explain how you got it to work?

manivalaguru, 2021-05-03:
i have done this change  in LoraMac.cpp
in the funcion LORA_Cycle(sBuffer *Data_Tx, sBuffer *Data_Rx, RFM_command_t *RFM_Command, sLoRa_Session *Session_Data,
 									sLoRa_OTAA *OTAA_Data, sLoRa_Message *Message_Rx, sSettings *LoRa_Settings)
	instead of EU_868
	#ifdef IN_865_867
		unsigned char previousChannelRX=LoRa_Settings->Channel_Rx;
		unsigned char previousDatarateRX=LoRa_Settings->Datarate_Rx;
		LoRa_Settings->Channel_Rx=CHRX2;
		LoRa_Settings->Datarate_Rx=SF10BW125;
		LoRa_Settings->Channel_Rx=previousChannelRX;
		LoRa_Settings->Datarate_Rx=previousDatarateRX;
	#endif
and in struct.h

#ifdef IN_865_867
  CHRX2 = 3,
i have done these changes kindly check and see if its works for you too

SiddhantArya, 2021-05-03:
Oh..ok..thanks. I'll make the required changes and update accordingly.

SiddhantArya, 2021-05-22:
Still can't get it to work @manivalaguru . 
The prompt is still 'Joining...'

manivalaguru, 2021-05-23:
can you give me your mail i can send my file

SiddhantArya, 2021-05-23:
siddhant.arya18@gmail.com

closed 2021-07-29 by stale[bot]