ElectronicCats/Beelan-LoRaWAN #61: How to check uplink confirmation?

postfalk, 2020-11-10:
I am trying to get an uplink confirmation by setting the flag on sendUplink. My network server (Chirpstack) acknowledges that it is confirmed uplink). How do I check in the code whether the packet has been received? I found the undocumented readAck method but it always returns true, even if I unplug the gateway and I am sure that there is no other gateway around. 

I am using the send-class-A-OTAA example modified like this:

```
void loop() {
  // Check interval overflow
  if(millis() - previousMillis > interval) {
    previousMillis = millis(); 

    sprintf(myStr, "Counter-%d", counter); 

    Serial.print("Sending: ");
    Serial.println(myStr);
    
    // change 1: requesting acknowledgement
    lora.sendUplink(myStr, strlen(myStr), 1, 1);
    counter++;
  }
  
  // change 2: try to get confirmation, always True even if gateway turned off and no messages in the network server
  ackStatus = lora.readAck();
  if (ackStatus) {
    Serial.println("Acknowledged");
  }
  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}
```

Mark-Wills, 2020-12-13:
Looking at the code, I note that `readAck()` checks the flag `Ack_Status` and only returns true if `Ack_Status` is true. However, looking further down in the code, it looks like `Ack_Status` is only set in the `update()` function:

```c++
 //LoRa cycle
      LORA_Cycle(&Buffer_Tx, &Buffer_Rx, &RFM_Command_Status, &Session_Data, &OTAA_Data, &Message_Rx, &LoRa_Settings);

      if ((Message_Rx.Frame_Control & 0x20) > 0)
        Ack_Status = NEW_ACK;
```
I have verified that the code is checking the correct bit in the correct location in the FHDR structure, and it is correct (bit weight 0x20).

Therefore, my best suggestion is to call `update()` just prior to calling `readAck()`:
```c++
lora.sendUplink(myStr, strlen(myStr), 1, 1);
lora.update();
if(readAck()) {
  // ack was confirmed by back-end
} else {
  // no confirmation received
}
```

I have checked the `sendUplink()` function, and it does _not_ call `update()` so maybe that is the issue? I defer to the wisdom of the maintainers, but it's worth a try!

Regards

Mark

geologic, 2021-01-26:
Any news about this issue? I would like to check for uplink confirmation too...

mr3188, 2021-02-01:
sendUplink() just enqueue the downlink to be sent in next update() loop, it doesn't actually send the uplink.
In the same way readData() returns already received (in the last update()) downlink data.

You can call update() before readAck() or you can check for ACK in the next Arduino loop()

postfalk, 2021-02-24:
I still have problems to get this to work. I changed the order and tried the class A example with the .readData() method but no luck.

Mark-Wills, 2021-02-25:
The issue might be that the network is not honouring the uplink
confirmation request. Might be worth checking with your network provider.

daeynasvistas, 2021-06-29:
> I still have problems to get this to work. I changed the order and tried the class A example with the .readData() method but no luck. I am using a Feather M0 Lora.

Im unable to get it to work also. 

what do you mean, the ".readData()" method?

closed 2021-07-06 by stale[bot]