ElectronicCats/Beelan-LoRaWAN #64: error:"validate dev-nonce error" have multiple nonce error .. then the device authenticates OK

daeynasvistas, 2020-11-13:
i have multiple errors if i reset the device and OTAA start again.

**error**
>    type:"OTAA"
>    error:"validate dev-nonce error"
>    fCnt:0

**error**
>    type:"OTAA"
>    error:"validate dev-nonce error"
>    fCnt:0

**error**
>    type:"OTAA"
>    error:"validate dev-nonce error"
>    fCnt:0

after some try/error the device are able to join and get new KEYS.
**join**
 **uplink**

but..? why the random nonce didnt get randomize everytime, in the LoRa_Send_JoinReq() 

**LoRa_Send_JoinReq(sLoRa_OTAA *OTAA_Data, sSettings *LoRa_Settings)**

```
    //Generate DevNonce
    Generate_DevNonce(OTAA_Data->DevNonce);
```

why are my new OTAA JoinRequest sending the old nonce? . or do the random generate a value that was already been used?

mr3188, 2020-11-13:
The problem with devNonce is that the state of the device at startup is usually the same so there is no way to generate a real random number. This lets the join process to work at first attempt but sometimes  several JoinRequests are needed until the server forget old values.

I think the random() function uses stdlib rand() function depending on the MCU implementation we get more or less random numbers. Some people recommends using an analogue floating pin value or battery level as random seed.

In future LoRaWAN releases devNonce will be replaced by a sequential value the node will need to store in persistent memory.

plachta-zabatt, 2020-11-13:
[image]

This is what we did to help with  that. It doesn't solve the problem for the same reason as mr3188 stated above, but it helps last a little longer. Every once and awhile I clear out the dev nonce history in (we're using chirpstack)

daeynasvistas, 2020-11-14:
ok.. 
so in the future devnonce will be increment and need to be persistent, is it possible to implement it?

is it possible to EEPROM the OTAA information .. and then reuse it? My device only need to reset once a day.. but what about the deep sleep ones? they will probably encounter devnonce error when "re OTAA" on waking up

or the error devnonce are not a problem when nodes are in place.. just when debugging ..like i see right now?.

there is some implementation on LMIC like this one: https://github.com/Edzelf/LoRa/blob/master/ESP_lora_tracker/ESP8266_loratracker.ino

may be good feature

daeynasvistas, 2020-11-16:
@plachta-zabatt .. Sorry.. at first i didnt understand your answer!

but i see now that you seed the random with the analog (i didnt see that at first).. and its mutch better! before the join only append after 4 o 5 trys (or even more some time, with join devnonce error)..

im using ESP-12F with only one analog (A0) .. and i have something connected.. but even then, this simple solution are very good.

**its mutch better now... thank!**

mr3188, 2020-11-16:
If you have a sensor connected to the A0 it can be even better as analog sensors don't use to return exact values and sometimes have  small fluctuations.

closed 2020-11-16 by daeynasvistas