ElectronicCats/Beelan-LoRaWAN #66: Problem sending messages

Samuel-ZDM, 2020-12-23:
Hello!

I have an Atmega328 with RF96. I am using the internal clock with 1 MHz.
I made the configuration with the following pins:

const sRFM_pins RFM_pins = {
  .CS = 10,
  .RST = 6,
  .DIO0 = 9,
  .DIO1 = 7,
  .DIO2 = 8,
  .DIO5 = 0xff,
};

I used a fixed channel, I believe it is as follows:

// Set Data Rate
  lora.setDataRate(SF7BW125);

  // set channel to random
  lora.setChannel(1);

My gateway is configured with SF7 and frequency 903.9.

I'm trying to check the serial output and it doesn't show anything, I don't know why that happens. I'm using an Arduino Uno without the microcontroller to analyze the output.

Thank you very much in advance!

sabas1080, 2020-12-23:
Hi @Samuel-ZDM Can you share you code or sketch?

Samuel-ZDM, 2020-12-23:
Thank you for your reply.

I'm using PlatformIo.

```
#include <Arduino.h>
#include <lorawan.h>

//ABP Credentials 
const char *devAddr = "260...";
const char *nwkSKey = "2989....";
const char *appSKey = "EA0D....";

const unsigned long interval = 10000;    // 10 s interval to send message
unsigned long previousMillis = 0;  // will store last time message sent
unsigned int counter = 0;     // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
  .CS = 10,
  .RST = 6,
  .DIO0 = 9,
  .DIO1 = 7,
  .DIO2 = 8,
  .DIO5 = 0xff,
};

void setup() {
  // Setup loraid access
  Serial.begin(9600);
  delay(2000);
  if(!lora.init()){
    Serial.println("RFM95 not detected");
    delay(5000);
    return;
  }

  // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_A);

  // Set Data Rate
  lora.setDataRate(SF7BW125);

  // set channel to random
  lora.setChannel(1);
  
  // Put ABP Key and DevAddress here
  lora.setNwkSKey(nwkSKey);
  lora.setAppSKey(appSKey);
  lora.setDevAddr(devAddr);
}

void loop() {
  // Check interval overflow
  if(millis() - previousMillis > interval) {
    previousMillis = millis(); 

    sprintf(myStr, "Counter-%d", counter); 

    Serial.print("Sending: ");
    Serial.println(myStr);
    
    lora.sendUplink(myStr, strlen(myStr), 0, 1);
    counter++;
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}
```

Samuel-ZDM, 2020-12-24:
I was thinking here.
My device has the MISO and MOSI pins of the LoRa module soldered on the Atmega328, could there be a problem?
Because I saw that the BeeleanMX library has some functions with SPI pins.
I may be completely wrong about this, but maybe these pins may be giving you trouble communicating?

mr3188, 2020-12-27:
SPI pins are used ro communicate with RFM96 module, you can share MISO, MOSI and SCLK pins with other SPI slaves but each one has to have its own slave select (CS or SS) pin. 

lora.Init() checks communications with FRM95 module.

Try something like this in setup()

`if(!lora.init()){
    Serial.println("RFM95 not detected");
    while(1);
  }

Serial.println("RFM95 detected");
...`

Check free memory also, if you are near the limit you can have strange problems and resets.

Samuel-ZDM, 2020-12-28:
Many thanks for the reply.

I will try to do this, because my device's serial communication does not work with this library. An interesting fact is that the same problem happens with the LMIC library, the TX RX serial communication does not work.

I use the SPIs pins of the microcontroller just to write the code on the device.

Samuel-ZDM, 2020-12-28:
The device cannot seem to send and is looped.
After going through the setup it shows the counter but it is not sending.

17: 56: 29.803 -> RFM95 detected
17: 56: 37.611 -> Sending: Counter-0
17: 57: 17.612 -> RFM95 detected
17: 57: 25.452 -> Sending: Counter-0

sabas1080, 2020-12-28:
Can you send us a schematic of your connection? @Samuel-ZDM

Samuel-ZDM, 2020-12-28:
Yes. Thank you.

[image]

Samuel-ZDM, 2020-12-29:
The microcontroller is working at a frequency of 1MHz, using the internal clock, can this work in the library's operation?

Another point I was able to verify is that when I put a string to be printed after the sendUplink function, loose characters are shown, nothing in relation to what should be printed.

wero1414, 2021-01-04:
If the device is starting the RFM95 and it try to send some data, it means that the SPI is working but when it tries to do Tx it doesnt respond through the DIO0 that the Tx has finished you should check the pinout.

wero1414, 2021-01-04:
closed by bad click sorry

Samuel-ZDM, 2021-01-05:
No problem!

I think the connections are correct, as I tested it with another library and the Uplink messages were sent.
The only library that I managed to make work on my device was this: https://github.com/ricaun/LoRaWanPacket 

I don't know if I can put the link here, but as a reference is the one above.

Thank you very much.

Samuel-ZDM, 2021-01-12:
[image]
In this image, it shows the messages in the serial, in which the device tries to send a message and then does not work again.
I tested the sending of messages with the microcontroller frequency of 1Mhz and 8Mhz of the internal clock.

Is the frequency giving problems in communication?

geologic, 2021-01-27:
Put a small delay after printing to Serial.
How are you powering your module? When transmitting it can use up to 100mA of current

Samuel-ZDM, 2021-02-02:
I am using two AA batteries to power the entire circuit. Tests with another library that I mentioned earlier worked.
In a recent research I did, I saw that the DIO0 pin of the LoRa module has to be connected to pin 2 of the arduino nano for example.
Could the pin I'm using be causing the problem?

Thanks a lot for the help

geologic, 2021-02-03:
I think you can use any pin you define on const sRFM_pins RFM_pins = { }
Pin2 of arduino nano is a INT pin, so maybe the example you saw use interrupts to react to events (my guess...)

Samuel-ZDM, 2021-04-27:
I did a search and managed to find that the DIO0 pin has to be on some interruption pin. So I switched to pin 3 of the Atmega328p but it still doesn't work. I tested it with the OTAA example. He is able to Joining but does not send, he locks up afterwards. What can this be?

The settings are the same, 1MHz of the micro, I'm not using the DIO5 pin.

closed 2021-07-05 by stale[bot]

Joso997, 2022-09-27:
Problem solved. You can't use 1 MHz, it barely works with 8Mhz internal. I recommend you get an external part for frequency.