ElectronicCats/Beelan-LoRaWAN #67: light sleep mode issue

Koraze, 2020-12-27:
Hi
I got an new issue (I also have minor issues but I will talk about these in another thread)

The lorawan library didn't work when ESP wakes up after a light sleep (`esp_light_sleep_start()`)
In fact, after a light sleep, the lora seems to stops working, so I have to reinit it with :

`
      esp_sleep_enable_timer_wakeup(esp_sleep);
      esp_light_sleep_start();

      SPI.begin(5,19,27,18);
      if(!lora.init()){
        Serial.println("LORA: lora module not detected");
        delay(5000);
        return;
      }

      lora.setDeviceClass(CLASS_A);
      lora.setDataRate(LORA_DATARATE);
      lora.setChannel(LORA_CHANNEL);
`

In that case, the lora succeed to send datas and receive acks, but my LoRaWAN receiver keeps telling me I got nothing...
I think it's because during `lora.init()`, all lora parameters (including actual session) are reset.

But all works again if I edit your library (lorawan-arduino-rfm.cpp > LoRaWANClass::init()) this way (value = 1 for full reset, value = 0 for half reset) :

`
bool LoRaWANClass::init(bool value)
{
	if(value)
	{
    // Lora Setting Class
    dev_class = CLASS_A;
    // Random seed
    randomSeed(analogRead(0));

    // Status
    RFM_Command_Status = NO_RFM_COMMAND;
    Rx_Status = NO_RX;
    Ack_Status = NO_ACK;

    // current channel
    currentChannel = MULTI;

    // Initialise session data struct (Semtech default key)
    memset(Address_Tx, 0x00, 4);
    memset(NwkSKey, 0x00, 16);
    memset(AppSKey, 0x00, 16);
    
    Frame_Counter_Tx = 0x0000;

    Session_Data.NwkSKey = NwkSKey;
    Session_Data.AppSKey = AppSKey;
    Session_Data.DevAddr = Address_Tx;
    Session_Data.Frame_Counter = &Frame_Counter_Tx;

    //Initialize OTAA data struct
    memset(DevEUI, 0x00, 8);
    memset(AppEUI, 0x00, 8);

    memset(AppKey, 0x00, 16);
    memset(DevNonce, 0x00, 2);
    memset(AppNonce, 0x00, 3);
    memset(NetID, 0x00, 3);
    OTAA_Data.DevEUI = DevEUI;
    OTAA_Data.AppEUI = AppEUI;
    OTAA_Data.AppKey = AppKey;
    OTAA_Data.DevNonce = DevNonce;
    OTAA_Data.AppNonce = AppNonce;
    OTAA_Data.NetID = NetID;
    
    // Device Class
    LoRa_Settings.Mote_Class = 0x00; //0x00 is type A, 0x01 is type C
    
    // Rx
#if defined(AS_923)
    LoRa_Settings.Datarate_Rx = 0x02;   //set to SF10 BW 125 kHz
#elif defined(EU_868)
    LoRa_Settings.Datarate_Rx = 0x03;   //set to SF9 BW 125 kHz
#else //US_915 or AU_915
    LoRa_Settings.Datarate_Rx = 0x0C;   //set to SF8 BW 500 kHz
#endif
    LoRa_Settings.Channel_Rx = 0x0A;    // set to recv channel

    // Tx
#if defined(US_915)
    LoRa_Settings.Datarate_Tx = drate_common = 0x02;   //set to SF7 BW 125 kHz
#elif defined(AU_915)
    LoRa_Settings.Datarate_Tx = drate_common = 0x02;   //set to SF7 BW 125 kHz
#else
    LoRa_Settings.Datarate_Tx = drate_common = 0x00;   //set to SF12 BW 125 kHz
#endif
    LoRa_Settings.Channel_Tx = 0x00;    // set to channel 0

    LoRa_Settings.Confirm = 0x00; //0x00 unconfirmed, 0x01 confirmed
    LoRa_Settings.Channel_Hopping = 0x00; //0x00 no channel hopping, 0x01 channel hopping

    // Initialise buffer for data to transmit
    memset(Data_Tx, 0x00, sizeof(Data_Tx));
    Buffer_Tx.Data = Data_Tx;
    Buffer_Tx.Counter = 0x00;
    // Initialise buffer for data to receive
    memset(Data_Rx, 0x00, sizeof(Data_Rx));
    Buffer_Rx.Data = Data_Rx;
    Buffer_Rx.Counter = 0x00;
    Message_Rx.Direction = 0x01; //Set down direction for Rx message
	}
    //Initialize I/O pins
    pinMode(RFM_pins.DIO0,INPUT);
    pinMode(RFM_pins.DIO1,INPUT);
    #ifdef BOARD_DRAGINO_SHIELD
    pinMode(RFM_pins.DIO5,INPUT);
    #endif
    pinMode(RFM_pins.DIO2,INPUT);
    pinMode(RFM_pins.CS,OUTPUT);
    pinMode(RFM_pins.RST,OUTPUT);
    
    digitalWrite(RFM_pins.CS,HIGH);
    
    // Reset
    digitalWrite(RFM_pins.RST,HIGH);
    delay(10);
    digitalWrite(RFM_pins.RST,LOW);
    delay(10);
    digitalWrite(RFM_pins.RST,HIGH);

    //Initialise the SPI port
    SPI.begin();
    
    /*** This prevents the use of other SPI devices with different settings ***/
    //SPI.beginTransaction(SPISettings(4000000,MSBFIRST,SPI_MODE0)); 

    //Wait until RFM module is started
    delay(50);
    
    //Initialize RFM module
    if(!RFM_Init()){
    return 0;
    }
    return 1;
}
`

Could you make your library works with light sleep (and maybe deep sleep) ?

My configuration : Heltec Lora32 v2 (ESP32 + sx1276)

sabas1080, 2020-12-27:
Hi @Koraze 

Can you test with this branch please https://github.com/BeelanMX/Beelan-LoRaWAN/pull/58? Ill wait for your comments

Koraze, 2020-12-28:
I will try when I will have some time (I don't know how to use branches so I need to make some researches before ;))
Hopefully, my solution works too !!

More details about my issue :

My setup 
 - LoraWAN End Node : Heltec Lora32 v2
 - LoraWAN Gateway : LPS8
 - LoraWAN Server : ChirpStack

My end node config
 - EU 868MHz, Class A, Mode OTAA, Channel 1, Data rate SF10BW125

My end node goal
 - Send data
 - Wait for ACK
 - Enter into light sleep mode during 7s
 - Repeat all

How to reproduce my issue
 - Init lora with lora.init()
 - Join my lorawan with lora.join()
 - Send new data then wait for ack (the lorawan server correctly receive my datas)
 - ...
 - Send new data again then wait for ack (the lorawan server still correctly receive my datas)
 - light sleep during 7s (in that mode, all radios including lora are disable)
 - Enable lora again with lora.init()
 - Send new data then wait for ack (the lorawan server shows nothing)
 - ...
 - Send new data again then wait for ack (the lorawan server will never shows anything again until I restart the end node)

sabas1080, 2020-12-28:
@Koraze use this link https://github.com/BeelanMX/Beelan-LoRaWAN/archive/DownlinkTimingFix.zip ;)

Koraze, 2020-12-29:
Great thanks !!

I just try your branch, I still have the same issue ... and I found another one ... :

 - After you init your lora then join a lorawan network, if you let the ESP32 takes a light sleep, you can't reach the network anymore (uplinks and downlinks). You have to re-init and join again !!
 - After you join a lorawan network, if you send uplinks, you will always get ack even if for some reasons that network falls some time later (gateway unplugged)

closed 2021-07-05 by stale[bot]