ElectronicCats/Beelan-LoRaWAN #70: OTAA not working

geologic, 2021-01-27:
Hi

I'm using your OTTA example to connect a SX1276+arduino to TTN. 
When running the sketch, it loops forever on "Joining..."

At TTN console i see thousands of join requests, it seems the node is flooding the gateway.
Does your library works on OTAA mode?

sabas1080, 2021-01-27:
Can you show us your code?

sabas1080, 2021-01-27:
this bug is working on #58

geologic, 2021-01-28:
So, it means this bug is fixed? But the solution is not in the repository, right?

mr3188, 2021-02-01:
If using EU868 you can try with #73

manivalaguru, 2021-06-10:
hello, i am using your library with nrf52832 and nice rf module which is sx1276 in 865-867 frewquency i can be able do class c and claas a in abp sucessfuly both uplinks and downlinks but while doing otta my code stucks on joining sometime it joins after few minutes sometime it never joins and downlinks also not happdening can you help me with this issue ?? i have already created an issue someone help me

sabas1080, 2021-07-09:
Fixed en release 2.0.0

closed 2021-07-09 by sabas1080