ElectronicCats/Beelan-LoRaWAN #71: Multiple uplink messages

geologic, 2021-01-27:
Hi

I'm using your ABP example to connect a SX1276+arduino to TTN.
I changed the code just to send one request, so i see at terminal "Sending: Counter-0"

At TTN console i see on that node thousands of uplink messages, all with payload "Counter-0", so i'm receiving lots of duplicated messages. It seems your library is flooding the gateway with messages.

Why is this happening?

sabas1080, 2021-01-27:
Hi @geologic 

Can you show us your code?

geologic, 2021-01-27:
Sure.
It is pretty much the same code of the example:

```
#include <lorawan.h>

//ABP Credentials 
const char *devAddr = "26015D3F";
const char *nwkSKey = "F67BB976CE405DC19F3FC0C4C6AB4CC9";
const char *appSKey = "CCF9DE27B57151410DC78AEF0B3FB089";

const unsigned long interval = 30000;    // 30 s interval to send message
unsigned long previousMillis = 0;  // will store last time message sent
unsigned int counter = 0;     // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

#define DEBUG_LED 7                   // D7 - led 
#define EN_LORA 2                     // D2 - enable LDO @ 3V3
const sRFM_pins RFM_pins = {
  .CS = 10,                           // RST - D9; CS - D10; MOSI - D11; MISO - D12; SCK - D13; DIO0 - D3; DIO1 - D4; DIO2 - D5; DIO5 - D6;
  .RST = 9,                           
  .DIO0 = 3,
  .DIO1 = 4,
  .DIO2 = 5,
  .DIO5 = 6,
};

void setup() {
  // Setup loraid access
  Serial.begin(57600);
  while(!Serial);

  pinMode(EN_LORA,OUTPUT);
  digitalWrite(EN_LORA,HIGH);

    pinMode(DEBUG_LED,OUTPUT);
    digitalWrite(DEBUG_LED,LOW);
  
  if(!lora.init()){
    Serial.println("LoRa module not found!");
    while (1) {}    
  } else {
    Serial.println("LoRa module found."); 
  }

  // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_A);

  // Set Data Rate
  //lora.setDataRate(SF9BW125);
  lora.setDataRate(SF7BW125);

  // set channel to random
  lora.setChannel(MULTI);
  
  // Put ABP Key and DevAddress here
  lora.setNwkSKey(nwkSKey);
  lora.setAppSKey(appSKey);
  lora.setDevAddr(devAddr);
}
  
void loop() {
  // Check interval overflow
  if(millis() - previousMillis > interval) {
    digitalWrite(DEBUG_LED,HIGH);
    previousMillis = millis(); 

    sprintf(myStr, "Counter-%d", counter); 

    Serial.print("Sending: ");
    Serial.println(myStr);
    
    lora.sendUplink(myStr, strlen(myStr), 0, 1);
    counter++;
     digitalWrite(DEBUG_LED,LOW);
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}
```

mr3188, 2021-02-01:
Hi @geologic
Which branch and region are you using?

geologic, 2021-02-02:
I'm using the master branch and region EU-868

mr3188, 2021-02-03:
I tried the same code and everything seems to work fine. The only part of your code I can't test is all related to EN_LORA pin.

It sends and un link every 30 seconds incrementing the count as expected.

wero1414, 2021-04-26:
Closing cause not activity

closed 2021-04-26 by wero1414