ElectronicCats/Beelan-LoRaWAN #81: how to see class c downlink messages

manivalaguru, 2021-04-28:
hello,
i a m using esp32 with sx1276 and using class c apb example i am sending the downlink messages in my application but could not see any thing on the console
recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
this is where i get the downlink messges right i want to know if my device works in class c or not can you help me with this
thanks and regards
manikandan v

halukmy, 2021-04-29:
can you downlink? @manivalaguru

manivalaguru, 2021-04-29:
no still trying my downlink messages coming to gateway but not to node i think i have to change some settings according to my region but i dnt know where exactly can some one help me

wero1414, 2021-05-07:
Hi!
There is one branch called downlink timming fixed you should try that to see the Class C donwlinks.

sabas1080, 2021-07-09:
Fixed en release 2.0.0

closed 2021-07-09 by sabas1080