ElectronicCats/Beelan-LoRaWAN #84: with nrf52840

manivalaguru, 2021-05-05:
does anyone have source code for this library with nrf52840 i want to intergrate with nrft52840 with sx1276
thanks and regards
manikandan v

sabas1080, 2021-05-06:
Try some of the examples in the folder examples, nrf52840 should work with the Arduino core and Mbed

manivalaguru, 2021-05-15:
hello,
i am porting this library to nrf52840 i can be able to do read and write in spi but i am getting stuck in this function
void RFM_Send_Package(sBuffer *RFM_Tx_Package, sSettings *LoRa_Settings)
{
  unsigned char i;
  unsigned char RFM_Tx_Location = 0x00;

  //Set RFM in Standby mode
  RFM_Switch_Mode(RFM_MODE_STANDBY);

  //Switch Datarate
  RFM_Change_Datarate(LoRa_Settings->Datarate_Tx);

  //Switch Channel
  RFM_Change_Channel(LoRa_Settings->Channel_Tx);

  //Switch DIO0 to TxDone
  RFM_Write(RFM_REG_DIO_MAPPING1, 0x40);

  //Set IQ to normal values
  RFM_Write(RFM_REG_INVERT_IQ,0x27);
  RFM_Write(RFM_REG_INVERT_IQ2,0x1D);

  //Set payload length to the right length
  RFM_Write(RFM_REG_PAYLOAD_LENGTH,RFM_Tx_Package->Counter);

  //Get location of Tx part of FiFo
  RFM_Tx_Location = RFM_Read(0x0E);

  //Set SPI pointer to start of Tx part in FiFo
  RFM_Write(RFM_REG_FIFO_ADDR_PTR, RFM_Tx_Location);
 int data=0;
  //Write Payload to FiFo
  for (i = 0;i < RFM_Tx_Package->Counter; i++)
  {
 
    RFM_Write(RFM_REG_FIFO, RFM_Tx_Package->Data[i]);

   NRF_LOG_INFO("%d",i);
   NRF_LOG_FLUSH();
### _****_
  }
//Switch RFM to Tx
  RFM_Write(RFM_REG_OP_MODE,0x83);

  //Wait for TxDone
 // while(digitalRead(RFM_pins.DIO0) == LOW);
_while (nrf_gpio_pin_read(DIO0) == 0);_
  //Clear interrupt
  RFM_Write(RFM_REG_IRQ_FLAGS,0x08);

  //Switch RFM back to receive if it is a type C mote
  if(LoRa_Settings->Mote_Class == CLASS_C)
  {
    //Switch Back to Continuous receive
    RFM_Continuous_Receive(LoRa_Settings);
  }
}
in this line while(nrf_gpio_pin_read(DI00)==0))
can you tell me the reason for this
thanks and regards
manikandan v

sabas1080, 2021-05-15:
//Wait for TxDone
// while(digitalRead(RFM_pins.DIO0) == LOW);

wait for the transmission in the RFM95 or SX1272 to finish

manivalaguru, 2021-05-15:
so if it is does not get cleared means my uplink does not happened properly right can you tell me the factors that affecting this my program getting stuck at this line and i dont know how tp debug this issue can you help me
thanks and regards
manikandan v

sabas1080, 2021-06-13:
Yes, nothing is being received from the DIO0 pin, you must check connections and the pin is correctly declared

closed 2021-07-09 by sabas1080