ElectronicCats/Beelan-LoRaWAN #85: Use RX2 other then SF12BW125 ? for DOWNLINK

daeynasvistas, 2021-05-09:
Can i change RX2 from default SF12BW125 to another SF?

it seems if i change to let say SF8BW125, nodes never get the packet
**do BeeLan force 1.0X LoRaWAN Regional Parameters to SF12BW125 ?**

**problem:**
With SF7->SF12 i can use OTAA and all are OK.
using SF12BW125 for DOWNLINK on RX2 all OK.

change RX2 to another SF for DOWNLINK, and nodes never gets packet.

daeynasvistas, 2021-05-09:
OK .. it seems its allways listen to SF12 .. 

```
*****************************************************************************************
* Description : Function to switch RFM to continuous receive mode, used for Class C motes
*
* Arguments   : *LoRa_Settings pointer to sSettings struct
*****************************************************************************************
*/
void RFM_Continuous_Receive(sSettings *LoRa_Settings)
{
  //Change DIO 0 back to RxDone and DIO 1 to rx timeout
  RFM_Write(RFM_REG_DIO_MAPPING1,0x00);

  //Invert IQ Back
  RFM_Write(RFM_REG_INVERT_IQ, 0x67);
  RFM_Write(RFM_REG_INVERT_IQ2, 0x19);
  
	//Change Datarate and channel.
  // This depends on regional parameters
#ifdef EU_868
  RFM_Change_Datarate(SF12BW125);
  RFM_Change_Channel(CHRX2);
#else
	RFM_Change_Datarate(LoRa_Settings->Datarate_Rx);
	RFM_Change_Channel(LoRa_Settings->Channel_Rx);
#endif

	//Switch to continuous receive
	RFM_Switch_Mode(RFM_MODE_RXCONT);
}

```

so if i change 

```
#ifdef EU_868
  RFM_Change_Datarate(SF12BW125);
  RFM_Change_Channel(CHRX2);
#else
```
to
```
#ifdef EU_868
  RFM_Change_Datarate(SF8BW125);
  RFM_Change_Channel(CHRX2);
#else
```

 .. the node can get packet. When RX2 of NS are also SF8BW125.

is the ADR implementation job, to change that? can ADR also change the Listen SF?

daeynasvistas, 2021-05-09:
i can see in the gateway:

```
"FCnt": 10,
            "FOpts": [
                {
                    "CID": "RXParamSetupReq",
                    "RX1DRoffset": 0,
                    "RX2DataRate": 2,
                    "RX2Frequency": 869525000
                }
            ]

```

is this the MAC command "RXParamSetupReq" to change RX? on the node?

sabas1080, 2021-07-09:
Check release 2.0.0

closed 2021-09-15 by stale[bot]