ElectronicCats/Beelan-LoRaWAN #89: EU868 + OTAA + Class C Downlink

gigabyte23, 2021-06-25:
Hello!
I'm using yours OTAA examples on RAK4260 (SAMR34J18 + SX1276) /w BastWAN bootloader - EU868 Band.
Everything works ok until I stay in Class A. (both Uplink+Downlink+ACK... etc.)

When I switch to Class C, I can join network using OTAA, send packets to my GW but there is no downlink to my LoRaWAN node. Next packets are queued on my gateway but cannot reach my node. I'm using lora.readData(); and lora.update(); to check if there is new data available.

Are there any additional settings in library to set RX or mote to Class C (0x00 to 0x01)?

lukhof, 2021-06-30:
Hey,
i have the same problem. I tried the DownlinkTimingFix branch but i can not receive downlinks for the EU868 band...
Does anyone has a working solution or a hint what to change?

daeynasvistas, 2021-06-30:
did your NS server send downlink with SF12BW125 ?

#ifdef EU_868
  RFM_Change_Datarate(SF12BW125);
  RFM_Change_Channel(CHRX2);
#else

https://github.com/BeelanMX/Beelan-LoRaWAN/issues/85

daeynasvistas, 2021-06-30:
> Edit:
> i got a downlink in class C mode on the master branch! it was my/the GW fault. I scheduled a downlink in the GW but it did send the downlink after an uplink only....

With OTAA you need to send "first uplink" (at least with ChirpStack) for the autentication process be completed.
You are out of sync if you try to send payload befor the "First Uplink" (fcnt 0) from node

lukhof, 2021-06-30:
Yes i did not send the first uplink. It is my first time using LoRa and wasn't aware of it. Thank you! Now it's working flawlessly!

gigabyte23, 2021-06-30:
Not working in My case - I'm able to authenticate using first uplink but still cannot read any next downlink packets from gateway.

daeynasvistas, 2021-06-30:
> Not working in My case - I'm able to authenticate using first uplink but still cannot read any next downlink packets from gateway.

do you use private NS server like Chirpstack? or TTN?

i think TTN use SF9 for RX2

gigabyte23, 2021-07-01:
I'm using Multitech Conduit Gateway when there is possible to generate packets using NodeRED or even injecting raw data to downlink queue.

daeynasvistas, 2021-07-02:
> I'm using Multitech Conduit Gateway when there is possible to generate packets using NodeRED or even injecting raw data to downlink queue.
> 
> Also tested with different SF's, RX0, 1 and 2 with no success.

  RFM_Change_Datarate(SF12BW125);
  RFM_Change_Channel(CHRX2);

did you test with RX2 SF12BW125

sabas1080, 2021-07-09:
Check release 2.0.0

closed 2021-09-15 by stale[bot]