ElectronicCats/Beelan-LoRaWAN #92: Confirmed uplink - Class A

dhtek, 2021-07-16:
Hello,

It seems the ReadAck never catch the ack.

I got the issue with previous version but after upgraded to V 2.0.0, still have the same problem.

The simple code: 

`lora.sendUplink(myStr, strlen(myStr), 1, 1);`
` ackStatus = lora.readAck();
  if (ackStatus) {
    Serial.println("Acknowledged");
  }`

```
  // Check Lora RX
  lora.update();
```

Class A - TTN V3

gnoubi, 2021-07-29:
Dear, 
I have the same problem with the same configuration (Class A - TTN V3 - EU868)

Do you have any solution about this problem ? 

best regard

gnoubi, 2021-09-27:
hi,

Do you have any response to this problem? Today, I am blocked

Cheers

lukhof, 2021-09-29:
You have to call lora.update() before lora.readAck(), because lora.sendUplink(...) only fills the TX buffer. LoRa transmission and receive happens in lora.update() only. Does this help?

Best regards

closed 2021-12-05 by stale[bot]