ElectronicCats/Beelan-LoRaWAN #102: Class C Downlink over OTAA (TTN)

fanzypantz, 2021-10-06:
From what I've read on other issues, class C downlink issues should have been solved in 2.0? However, with my device, using an RFM95x chip on ESP32 I'm only able to receive downlink messages using ABP, not over OTAA. It would obviously be preferable to use OTAA.

I'm wondering if there is something wrong with my code or if it could be buggy? I've tried SF7, SF9 and SF12, and the most reliable seems to have been SF7, that is to join TTN. None of them gave me a downlink message. 

```c++
#include <Arduino.h>
#include <lora.h>
#include <lorawan.h>

// OTAA credentials
const char *devEui = "";
const char *appEui = "";
const char *appKey = "";

const unsigned long loraInterval = 20 * 1000;    // 10 s interval to send message
unsigned long loraPreviousMillis = 0;  // will store last time message sent
unsigned int loraCounter = 0;     // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
  .CS = 5,
  .RST = 14,
  .DIO0 = 2,
  .DIO1 = 12,
  .DIO2 = -1,
  .DIO5 = -1,
};

void setupLoRa() {
    if(!lora.init()){
        Serial.println("[LoRa] RFM95 not detected");
        delay(5000);
        return;
    }

    // Set LoRaWAN Class change CLASS_A or CLASS_C
    lora.setDeviceClass(CLASS_C);

    // Set Data Rate
    lora.setDataRate(SF7BW125);

    // set channel to random
    lora.setChannel(MULTI);
    
    // Put OTAA Key and DevAddress here
    lora.setDevEUI(devEui);
    lora.setAppEUI(appEui);
    lora.setAppKey(appKey);

    // Join procedure
    bool isJoined;
    do {
        Serial.println("[LoRa] Joining over OTAA...");
        isJoined = lora.join();

        //wait for 10s to try again
        delay(10000);
    }while(!isJoined);
    Serial.println("[LoRa] Joined a network");
}

void LoRaLoop() {
  
  // Check interval overflow
  if(millis() - loraPreviousMillis > loraInterval) {
    loraPreviousMillis = millis(); 
    Serial.println("[LoRa] Sending lora message");

    sprintf(myStr, "Counter-%d", loraCounter); 

    
    lora.sendUplink(myStr, strlen(myStr), 0, 1);
    loraCounter++;
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.print("====>> ");
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}
```

closed 2021-12-09 by fanzypantz

mirhamza708, 2024-10-04:
did you fix the issue fanzypantz?
can you please share if it was fixed because I am having the same issue.

fanzypantz, 2024-10-04:
> did you fix the issue fanzypantz? can you please share if it was fixed because I am having the same issue.

Hi, I don't think I ever managed to get proper Class C communication using the existing easy to use libraries to work at that time. And we moved away from using LoRaWan. Good luck.