ElectronicCats/Beelan-LoRaWAN #106: Node is hanging in Join proccedure

franciscoqqq, 2021-11-17:
Hi!

We are having a problem in our node that is hanging in the joining proccedure.
We are using a STM32F103C8 with a RFM95W module.
STM32duino core. Library: [Beelan-LoraWan](https://github.com/BeelanMX/Beelan-LoRaWAN/).
Frequency plan: AU915
GW: RAK2245 (Chirpstack GatewayBridge+Network Server+App Server all runnning in the same Rpi).

The problem happens randomly after restarting the node by software, and trying to do the join process. We tried to use timeout interrupts within the join process and it was not useful either. 
It is as if the micro was completely left frozen even the timers.

I attached images of the JoinRequest payload and the payload error and a script with the main code. 
[image]
[image]
[image]
[main](https://drive.google.com/file/d/1XxzYuGdO0rNUjSHUNYWGTFZfq2qZBvMA/view?usp=sharing)

I´m really stuck and I don´t know how to proceed.

Thanks a lot in advance!

FQ

Parradan, 2022-01-17:
The problem was solved, it could not be detected because the error occurred, but it was possible to prevent it from happening. Simply switch from SF7 to any other SF and the problem goes away.

a-backman, 2022-01-31:
I think that I'm having the same issue. The node hangs on lora.join(), it just prints "Joining...". I've confirmed that the join request is sent and the server replies with a join accept. 

If i poke at the antenna connector it sometimes comes out of the freeze and prints "Join request sent". If i poke some more it will eventually join.

I'm using a LILYGO TTGO LORA32 module, EU868 band. 

`#define EU_868

#include <lorawan.h>

// OTAA credentials
const char *devEui = "";
const char *appEui = "";
const char *appKey = "";

const unsigned long interval = 10000; // 10 s interval to send message
unsigned long previousMillis = 0;	  // will store last time message sent
unsigned int counter = 0;			  // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
	.CS = 18,
	.RST = 14,
	.DIO0 = 26,
	.DIO1 = 35,
	.DIO2 = 34,
	.DIO5 = -1,
};

void setup()
{
	Serial.begin(115200);

	if (!lora.init())
	{
		Serial.println("RFM95 not detected");
		delay(5000);
		return;
	}

	// Set LoRaWAN Class change CLASS_A or CLASS_C
	lora.setDeviceClass(CLASS_C);

	// Set Data Rate
	lora.setDataRate(SF9BW125);

	// set channel to random
	lora.setChannel(MULTI);

	// Put OTAA Key and DevAddress here
	lora.setDevEUI(devEui);
	lora.setAppEUI(appEui);
	lora.setAppKey(appKey);

	// Join procedure
	bool isJoined;
	do
	{
		Serial.println("Joining...");
		isJoined = lora.join();
		Serial.println("Join request sent");
		//wait for 10s to try again
		delay(1000);
	} while (!isJoined);
	Serial.println("Joined to network");
}

void loop()
{

	// Check interval overflow
	if (millis() - previousMillis > interval)
	{
		previousMillis = millis();

		sprintf(myStr, "Counter-%d", counter);

		Serial.print("Sending: ");
		Serial.println(myStr);

		lora.sendUplink(myStr, strlen(myStr), 0, 1);
		counter++;
	}

	recvStatus = lora.readData(outStr);
	if (recvStatus)
	{
		Serial.println(outStr);
	}

	// Check Lora RX
	lora.update();
}
`

closed 2022-04-09 by stale[bot]