ElectronicCats/Beelan-LoRaWAN #108: Issues accepting join message from TTN - OTAA

fanzypantz, 2021-12-09:
I'm not sure why but sometimes the system won't accept the join message even though the TTN network is getting the OTAA join messages and sending the join-accept message. I'm on the european frequency. What could be the cause for this issue? 

```c++
#include <Arduino.h>
#include <lora.h>
#include <lorawan.h>
#include <LoraMessage.h>
#include <relay.h>
#include <states.h>

// OTAA credentials
const char *devEui = "";
const char *appEui = "";
const char *appKey = "";

const unsigned long loraInterval = 20 * 1000;    // 10 s interval to send message
unsigned long loraPreviousMillis = 0;  // will store last time message sent
unsigned int loraCounter = 0;     // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
  .CS = 5,
  .RST = 14,
  .DIO0 = 2,
  .DIO1 = 12,
  .DIO2 = -1,
  .DIO5 = -1,
};

void sendLoRaMessage(int state) {
  Serial.println("[LoRa] Sending lora message");

  sprintf(myStr, "Counter-%d", loraCounter); 

  LoraMessage message;
  message.addUint8(state);
  if(state != STATE_WAKEUP || state != STATE_HEARTBEAT) {
    message.addRawFloat(meterData[0].lastReading);        
    message.addRawFloat(meterData[0].powerDraw);        
    message.addRawFloat(meterData[1].lastReading);
    message.addRawFloat(meterData[1].powerDraw);
    message.addRawFloat(meterData[2].lastReading);
    message.addRawFloat(meterData[2].powerDraw);
    message.addRawFloat(meterData[3].lastReading);
    message.addRawFloat(meterData[3].powerDraw);
  }

  lora.sendUplink((char *) message.getBytes(), message.getLength(), 0, 1);
  loraCounter++;
}

void setupLoRa() {
    if(!lora.init()){
        Serial.println("[LoRa] RFM95 not detected");
        delay(5000);
        return;
    }

    // Set LoRaWAN Class change CLASS_A or CLASS_C
    lora.setDeviceClass(CLASS_C);

    // Set Data Rate
    lora.setDataRate(SF12BW125);

    // set channel to random
    lora.setChannel(MULTI);
    
    // Put OTAA Key and DevAddress here
    lora.setDevEUI(devEui);
    lora.setAppEUI(appEui);
    lora.setAppKey(appKey);

    // Join procedure
    bool isJoined;
    do {
        Serial.println("[LoRa] Joining over OTAA...");
        isJoined = lora.join();

        //wait for 10s to try again
        delay(10000);
    }while(!isJoined);
    Serial.println("[LoRa] Joined a network");

    // Send inital wakeup message
    sendLoRaMessage(STATE_WAKEUP);
}

void LoRaLoop() {

  if(     
    relayStatus[0] != false || 
    relayStatus[1] != false || 
    relayStatus[2] != false || 
    relayStatus[3] != false
  ) {
    if(millis() - loraPreviousMillis > loraInterval) {
      loraPreviousMillis = millis(); 
      sendLoRaMessage(STATE_METER);
    }
  } else {
    if(millis() - loraPreviousMillis > loraInterval) {
      loraPreviousMillis = millis(); 
      sendLoRaMessage(STATE_HEARTBEAT);
    }
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.print("[LoRa] Response: ");
    // for (auto& byte : outStr) {
    //     Serial.printf("%02X ", byte);
    // }
    // Serial.println("");
    String message = String(outStr);
    Serial.println(message);

    recvStatus = 0;
  }

  
  // Check Lora RX
  lora.update();
}
```

negis02, 2022-01-02:
I have same issue, but i try to join OTAA in Helium network, i can monitor traffic with HackRF and see, that in Helium gateways always responding (Join Accept) in same frequency, but in Helium console i see that should respond in same frequency channel like was requested to join. My code is similar, STM32F103, EU868, LoRa1276F30 G-NiceRF

negis02, 2022-01-08:
Actually everything working for my, just look like downlink join accept message should be strong enough to receive with my devise. I should drive more close to city to  successfully receive join accept message.

aturowsk, 2022-02-04:
I had a similar issue, and it took many attempts to join. 
I have now solved the problem and connect in one join request. I changed the Data rate to SF8BW125 and used the new wakeUp() method. The wakeUp method is not released in a version yet, so you need to download manually.

Am am using the following parameters
  #define EU_868
 lora.setDeviceClass(CLASS_A);
 lora.setDataRate(SF8BW125);
 lora.setChannel(MULTI);
 lora.wakeUp();

Make sure you have pins DIO0 and DIO1 connected too.

Also add some code to show the channel. 

`Serial.println("Waking up LoRa chip ");
  lora.wakeUp();
  bool isJoined;
  do {
    Serial.print("Joining...on channel: "); Serial.println(lora.getChannel());
    isJoined = lora.join();
    delay(10000);
  } while(!isJoined);
  Serial.print("Joined to network: "); Serial.println(lora.getChannel());`

I hope this helps others as this has taken me a long time to track down.

envyst, 2022-04-08:
> I had a similar issue, and it took many attempts to join. I have now solved the problem and connect in one join request. I changed the Data rate to SF8BW125 and used the new wakeUp() method. The wakeUp method is not released in a version yet, so you need to download manually.
> 
> Am am using the following parameters #define EU_868 lora.setDeviceClass(CLASS_A); lora.setDataRate(SF8BW125); lora.setChannel(MULTI); lora.wakeUp();
> 
> Make sure you have pins DIO0 and DIO1 connected too.
> 
> Also add some code to show the channel.
> 
> ` Serial.println("Waking up LoRa chip ");
> 
> lora.wakeUp(); bool isJoined; do { Serial.print("Joining...on channel: "); Serial.println(lora.getChannel()); isJoined = lora.join(); delay(10000); } while(!isJoined); Serial.print("Joined to network: "); Serial.println(lora.getChannel());`
> 
> I hope this helps others as this has taken me a long time to track down.
> 
> This post helped me too for changing the Data Rate. [#106 (comment)](https://github.com/BeelanMX/Beelan-LoRaWAN/issues/106#issuecomment-1014427902)

where can I download the wake up method?

closed 2022-06-14 by stale[bot]