ElectronicCats/Beelan-LoRaWAN #115: Downlink not working

a-backman, 2022-03-07:
Hi,

I have trouble getting downlink to work.
I'm using this device: https://www.makerfabs.com/wiki/index.php?title=LoRa_AC_Dimmer.
Joining and uplink works fine. I've tried sending a downlink package and I can see that my network server (chirpstack) sends it, but it is not received by the device.

After a while I get garbage output:
```
Joining...
Joined to network
Sending: Counter-0
Sending: Counter-1
Sending: Counter-2
Sending: Counter-3
Sending: Counter-4
Sending: Counter-5
Sending: Counter-6
Sending: Counter-7
Sending: Counter-8
Sending: Counter-9
Sending: Counter-10
====>

Sending: Counter-11
====>
�ăPc|w{�ko�0␁g+�׫vʂ�}�YG�Ԣ���r����&6?��4���q�1␕␄�#�␘�␅�␇␒���'�u	�,␚␛nZ�R;ֳ)�/�S�
Sending: Counter-12
====>

Sending: Counter-13
====>
�4���q�1␕␄�#�␘�␅�␇␒���'�u	�,␚␛nZ�
Sending: Counter-14
====>
۽M
Sending: Counter-15
====>
s��␁&����QN�PR{9i��-����{��Z��␇t␇/��

```

My code:
```
#define EU_868

#include <lorawan.h>

// OTAA credentials
const char *devEui = "0000000000000000";
const char *appEui = "0000000000000000";
const char *appKey = "00000000000000000000000000000000";

const unsigned long interval = 10000; // 10 s interval to send message
unsigned long previousMillis = 0;	  // will store last time message sent
unsigned int counter = 0;			  // message counter

char myStr[50];
char outStr[255];

byte recvStatus = 0;

bool ackStatus;

const sRFM_pins RFM_pins = {
	.CS = 10,
	.RST = 9,
	.DIO0 = 2,
	.DIO1 = 6,
	.DIO2 = 7,
	.DIO5 = 8
};

void setup()
{
	// Setup loraid access
	Serial.begin(9600);
	while (!Serial);
	if (!lora.init())
	{
		Serial.println("RFM95 not detected");
		delay(5000);
		return;
	}

	// Set LoRaWAN Class change CLASS_A or CLASS_C
	lora.setDeviceClass(CLASS_C);

	// Set Data Rate
	lora.setDataRate(SF9BW125);

	// set channel to random
	lora.setChannel(MULTI);

	// Put OTAA Key and DevAddress here
	lora.setDevEUI(devEui);
	lora.setAppEUI(appEui);
	lora.setAppKey(appKey);

	// Join procedure
	bool isJoined;
	do
	{
		Serial.println("Joining...");
		isJoined = lora.join();

		// wait for 10s to try again
		delay(10000);
	} while (!isJoined);
	Serial.println("Joined to network");
}

void loop()
{
	// Check interval overflow
	if (millis() - previousMillis > interval)
	{
		previousMillis = millis();

		sprintf(myStr, "Counter-%d", counter);

		Serial.print("Sending: ");
		Serial.println(myStr);

		lora.sendUplink(myStr, strlen(myStr), 0, 1);
		counter++;
	}

	recvStatus = lora.readData(outStr);
	if (recvStatus)
	{
		Serial.println("====>");
		Serial.println(outStr);
	}

	// Check Lora RX
	lora.update();
}
```

CCMDD, 2022-03-15:
> Hi,
>
> I have trouble getting downlink to work. I'm using this device: https://www.makerfabs.com/wiki/index.php?title=LoRa_AC_Dimmer. Joining and uplink works fine. I've tried sending a downlink [...]

same problem here on 2.0 and 1.5.4 version, default c class abp sketch

CCMDD, 2022-03-15:
41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59
A   B   C   D   E   F   G   H   I    P  Q   R   S   T   U   V   W  X   Y

a-backman, 2022-03-15:
I've updated my sketch to print the received data as HEX instead. I still cannot make sense out of what I receive. Sometimes I get this garbage data directly after join, sometimes it takes a while.

It does not correspond with when I send the downlink message. The downlink message I've tested with is "ABAB" (0x41, 0x42, 0x41, 0x42).

```
Joining...
Joined to network
Sending: Counter-0
[]
Sending: Counter-1
[FFFFFFD0FFFFFFEFFFFFFFAAFFFFFFFB434D33FFFFFF8545FFFFFFF927F503CFFFFFF9FFFFFFFA851FFFFFFA340FFFFFF8FFFFFFF92FFFFFF9D38FFFFFFF5FFFFFFBCFFFFFFB6FFFFFFDA2110FFFFFFFFFFFFFFF3FFFFFFD2FFFFFFCDC13FFFFFFEC5FFFFFFF974417FFFFFFC4FFFFFFA77E3D645D197360FFFFFF814FFFFFFFDC222AFFFFFF90FFFFFF8846FFFFFFEEFFFFFFB814FFFFFFDE5EBFFFFFFDBFFFFFFE0323AA496245CFFFFFFC2FFFFFFD3FFFFFFAC62FFFFFF91FFFFFF95FFFFFFE479]
Sending: Counter-2
[676F794620FFFFFFF5FFFFFF8AFFFFFF9AFFFFFFABBFFFFFFA41C3012FFFFFFB37010FFFFFF8E21397]
Sending: Counter-3
[]
Sending: Counter-4
[1B6E5AFFFFFFA0523BFFFFFFD6FFFFFFB329FFFFFFE32FFFFFFF8453FFFFFFD1]
Sending: Counter-5
[29FFFFFFB0FFFFFFF2FFFFFF953462FFFFFFB41822FFFFFF92FFFFFFFDDFFFFFF8022FFFFFFC2FFFFFFFD60732326FFFFFFD27FFFFFFFDCFFFFFFD9FFFFFFD72368FFFFFF94FFFFFFF332FFFFFF84FFFFFF97FFFFFFE267FFFFFFE16]
Sending: Counter-6
[722D36]
Sending: Counter-7
[8668568466FFFFFF9D1E167431EFFFFFF95FFFFFFBD1CFFFFFF9CFFFFFFBBFFFFFFD8D56FFFFFFEB3744FFFFFFE3161FFFFFFD036FFFFFFBC1D]
Sending: Counter-8
[]
Sending: Counter-9
[FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFC2FFFFFFEDFFFFFFE97C34637C777BFFFFFFF26B6FFFFFFFC5301672BFFFFFFFEFFFFFFD7FFFFFFAB76FFFFFFCAFFFFFF82FFFFFFC97DFFFFFFFA5947FFFFFFF0FFFFFFADFFFFFFD4FFFFFFA2FFFFFFAFFFFFFF9CFFFFFFA472FFFFFFC0FFFFFFB7FFFFFFFDFFFFFF9326363FFFFFFFF7FFFFFFCC34FFFFFFA5FFFFFFE5FFFFFFF171FFFFFFD831154FFFFFFC723FFFFFFC318FFFFFF965FFFFFF9A712FFFFFF80FFFFFFE2FFFFFFEB27FFFFFFB2759FFFFFF832C1A1B6E5AFFFFFFA0523BFFFFFFD6FFFFFFB329FFFFFFE32FFFFFFF8453FFFFFFD1]
Sending: Counter-10
[BFFFFFFDBFFFFFFE0323AA496245CFFFFFFC2FFFFFFD3FFFFFFAC62FFFFFF91FFFFFF95FFFFFFE479FFFFFFE7FFFFFFC8376DFFFFFF8DFFFFFFD54EFFFFFFA96C56FFFFFFF4FFFFFFEA657AFFFFFFAE8FFFFFFBA78252E1CFFFFFFA6FFFFFFB4FFFFFFC6FFFFFFE8FFFFFFDD741F4BFFFFFFBDFFFFFF8BFFFFFF8A703EFFFFFFB566483FFFFFFF6E613557FFFFFFB9FFFFFF86FFFFFFC11DFFFFFF9EFFFFFFE1FFFFFFF8FFFFFF981169FFFFFFD9FFFFFF8EFFFFFF94]
Sending: Counter-11
[696E672E2E2E]
Sending: Counter-12
[FFFFFFD5FFFFFFBE3FFFFFFFD3FFFFFFC7FFFFFFBBFFFFFFD8FFFFFFDBFFFFFFF7FFFFFFBFFFFFFFFBFFFFFFC7FFFFFFA1FFFFFFBEFFFFFFFFFFFFFFD655FFFFFFEE6F74FFFFFFFAFFFFFFE3FFFFFFAB6E4E63FFFFFFF67D7DFFFFFFE4FFFFFFE77E5E5EFFFFFF9D51FFFFFFEBFFFFFFD633FFFFFFDB7F6DFFFFFFAF77FFFFFFFFFFFFFFB1FFFFFFF627FFFFFFDAFFFFFFDEFFFFFFDB4EBFFFFFFD88F854BFFFFFFD88BFFFFFFD81A82516F8FFFFFFB2FFFFFFE73318C8C638FFFFFFA4FFFFFF8AFFFFFFF8FFFFFFDAFFFFFFA1FFFFFFBAFFFFFFD03DFFFFFFE4FFFFFF9FFFFFFFD360FFFFFFB2FFFFFFE7331FFFFFFEDFFFFFF99761EFFFFFFC557FFFFFFD6133474B6EFFFFFFF8FFFFFFCE4FFFFFFFE2FFFFFFD2FFFFFFF76D3369FFFFFFF6FFFFFFBBFFFFFF948458768668568466FFFFFF9D1E167431EFFFFFF95FFFFFFBD1CFFFFFF9CFFFFFFBBFFFFFFD8D56FFFFFFEB3744FFFFFFE3161FFFFFFD036FFFFFFBC1D]
```

```
#define EU_868

#include <lorawan.h>

// OTAA credentials
const char *devEui = "cc50e3b5f67c40b4";
const char *appEui = "0000000000000000";
const char *appKey = "b59163edfcdad654267dd0469d8a55f5";

const unsigned long interval = 10000; // 10 s interval to send message
unsigned long previousMillis = 0;	  // will store last time message sent
unsigned int counter = 0;			  // message counter

char myStr[50];
char outStr[255];

byte recvStatus = 0;

bool ackStatus;

const sRFM_pins RFM_pins = {
	.CS = 10,
	.RST = 9,
	.DIO0 = 2,
	.DIO1 = 6,
	.DIO2 = 7,
	.DIO5 = 8};

void setup()
{
	// Setup loraid access
	Serial.begin(9600);
	while (!Serial)
		;
	if (!lora.init())
	{
		Serial.println("RFM95 not detected");
		delay(5000);
		return;
	}

	// Set LoRaWAN Class change CLASS_A or CLASS_C
	lora.setDeviceClass(CLASS_C);

	// Set Data Rate
	lora.setDataRate(SF9BW125);

	// set channel to random
	lora.setChannel(MULTI);

	// Put OTAA Key and DevAddress here
	lora.setDevEUI(devEui);
	lora.setAppEUI(appEui);
	lora.setAppKey(appKey);

	// Join procedure
	bool isJoined;
	do
	{
		Serial.println("Joining...");
		isJoined = lora.join();

		// wait for 10s to try again
		delay(10000);
	} while (!isJoined);
	Serial.println("Joined to network");
}

void loop()
{
	// Check interval overflow
	if (millis() - previousMillis > interval)
	{
		previousMillis = millis();

		sprintf(myStr, "Counter-%d", counter);

		Serial.print("Sending: ");
		Serial.println(myStr);

		lora.sendUplink(myStr, strlen(myStr), 0, 1);
		counter++;
	}

	recvStatus = lora.readData(outStr);
	if (recvStatus)
	{
		Serial.print("[");
		for (int n = 0; n <= 255; n++)
		{
			if (outStr[n] == 0)
				break;
			Serial.print(outStr[n], HEX);
		}
		Serial.println("]");
	}

	// Check Lora RX
	lora.update();
}
```

sascha-hemi, 2022-05-31:
Hey @a-backman did you get it working? I have the same issue :/

sascha-hemi, 2022-05-31:
My Output looks like:

`'�zV�,�S���1��_Qg;��J����e��s��d�+~���U��_�
���ϒ6����?���
������<�p�o�=;eZ�_ޙ�ں�K�}�5u1�t�HUМ�_������+y��e�0���K|�3=��1�&��^+����W����m��Z��|����a�qg�w^�f�]��[o��qQ���ۢ&M�Cs�>�B��]�z�)5c�����3;��j���Ξ�����4�~P���>�?w��N5ٚ�,{�8�N�6k��x��-㶆s�?��}�T��/�B?��?�@��Z󿽮 >�V���7�f�M�k�7�>�I$�:�/�X�����$�]��C.?@e{Ď�K},eeR�wy��BڟM                     M�ѧ��n?W��HB�;[�2/qdc�Q�&���������J��`

a-backman, 2022-06-02:
Hey @sascha-hemi, no I didn't. Still have the same issue.

sascha-hemi, 2022-06-03:
My Controller is an Sparkfun Pro Micro running at 8Mhz. Is 8Mhz to slow?

IoTThinks, 2022-07-10:
I created a pull request to fix the downlink in Class C for AS923/AS923-2 just now.
https://github.com/BeelanMX/Beelan-LoRaWAN/pull/126

wero1414, 2022-07-13:
@IoTThinks Merged already thanks!

closed 2022-07-13 by wero1414