ElectronicCats/Beelan-LoRaWAN #133: Arduino Nano reboot loop

mirkomatonti, 2022-07-26:
Hello, I'm trying to run the Class A ABP example. The problem is that my arduino nano keep rebooting itself after sending the first message.

Eric286, 2022-07-26:
Hello mirkomatonti!

To help us to understand your situation, we would like to ask you for the following additional information:
What Operating System and version are you using (e.g. Windows 11, macOS 12.0, Linux)?
What version of the Arduino IDE?
Did it work before? If so, which one was the last working sketch on the Arduino, circuit and power supply.
Please add a screenshot of the Device Manager of the computer (with “Ports” open). Please connect your board if using Windows.
Go to: Arduino IDE > File > Preferences > Show verbose output: Select both options: "Compilation" and "Upload". Then upload the Blink example (Arduino IDE > Files > Examples > Basics > Blink) and send us a copy of the whole text shown in the console.
Is your arduino nano original or did you buy it from a third party?

Have a nice day!

Kind Regards!

Eric Chavez, Support Team

Eric286, 2022-08-03:
Dear @mirkomatonti 

This is to remind you that your request is still pending and awaiting your feedback.

We would like to hear back from you about your previous request in order to know if the difficulty you were facing has been already solved or we can provide further assistance.

Thank you!

Have a good day!

Eric286, 2022-08-10:
Hello @mirkomatonti 

I will close this issue.
in case you face an issue, please contact us again so we help you to find the best solution.

closed 2022-08-10 by Eric286