ElectronicCats/Beelan-LoRaWAN #147: LoRaWAN protocol version?

sw-dev-code, 2022-11-25:
Can you tell me which LoRaWAN protocol versions are supported by this library?

IvanAmg, 2022-11-25:
Hello @sw-dev-code!
This library is based on version 1.0.3, it has been tested with version 1.0.4 and worked well, but some features may not be compatible or supported.
Let me know if I can help with anything else!

sw-dev-code, 2022-11-25:
@IvanAmg Thank you for your answer.
I have an ESP32-based device with SX1276 and I'm looking for a LoRaWAN Arduino library that supports 1.0.4 along with 1.0.3 and 1.0.2. 
Do you know which features are not compatible or missing, and do you have in plan to implement full support 1.0.4?

IvanAmg, 2022-11-28:
Not sure exactly which features are not supported, however, as this library is based on 1.0.3, all the new things on 1.0.4 should not be supported.
https://www.thethingsnetwork.org/article/whats-new-in-lorawan-104-1
Currently, we have a plan to support MAC commands.

IvanAmg, 2022-11-30:
@sw-dev-code 
Do you mind if we close this issue?
Or, is there anything else we can help with?

closed 2022-12-05 by IvanAmg