ElectronicCats/Beelan-LoRaWAN #149: LoRaWAN downlink not working

deepkap, 2023-01-13:
Uplink is working correctly, but not getting downlink on the esp32. 
We are using chirpstack 4.01 server.

xpeqex, 2023-01-13:
Hello there!

In order to help us to understand your situation, we would like to ask you for the following additional information:

- Which board are you using?

- Which IDE are you using to program the board?

- What example are you using?

xpeqex, 2023-01-18:
Hello @deepkap 

This is a message to remind you that your request has been pending for 5 days and awaits your comments. 
Our agent would like to hear from you about your previous request to see if the difficulty you were facing has been resolved or if we can provide further assistance. Otherwise, if we do not hear back from you within the next three days, the ticket will be closed.

Kind regards,
Electronic Cats Support Team

xpeqex, 2023-01-23:
Dear @deepkap 
Since we did not get any response from you and the time stated in the above message has passed, this issue will be closed.

Feel free to contact us again in case you have any problems with our products or libraries.

Have a nice day!

closed 2023-01-23 by xpeqex