ElectronicCats/Beelan-LoRaWAN #150: ElectronicCats

PadmaIoT, 2023-01-30:
Hello;
How are you? I am having a problem with hooking up the sx1276 chip-based Ai-thinker RA-01H module. I saw that the pinout and others parts of the program are formed with respect to Hope RF RFM95.

Can you please guide me or add support if needed?

wero1414, 2023-01-30:
It is fully compatible you just need to assign the correct pins to the actual connection on the radio

[image]

`
const sRFM_pins RFM_pins = {
  .CS = 20,
  .RST = 9,
  .DIO0 = 0,
  .DIO1 = 1,
  .DIO2 = 2,
  .DIO5 = 15,
};
`
On the pins that you connect them on your microcontroller

PadmaIoT, 2023-01-31:
Thanks for answering. I would definitely try it. One more thing to ask is for Class C do you have any support for OTAA? I did not find anything of class C with OTAA

Eric286, 2023-01-31:
Dear @PadmaIoT !

With OTAA, everything appears to be operating as it should.

Just keep in mind that there may be a bug, but you may notice it when debugging. It might be caused by your regional settings or anything similar.

In response to your query, it does support OTAA. 

Have a nice day!

Eric286, 2023-02-09:
Hello @PadmaIoT !

We will now close this issue.

However, if you have any doubts or questions, you can reopen the issue or if you think it is necessary, create another one.

Have an amazing day!

Kind regards!
Support Team.

PadmaIoT, 2023-02-13:
> Dear @PadmaIoT !
> 
> With OTAA, everything appears to be operating as it should.
> 
> Just keep in mind that there may be a bug, but you may notice it when debugging. It might be caused by your regional settings or anything similar.
> 
> In response to your query, it does support OTAA.
> 
> Have a nice day!

Hello Eric286,

Thanks for answering. In the example folder, I only see there is only ABP available for Class C Lorawan.  I asked  If there is any support we may get later as you are developing for OTAA Class C or not.

Eric286, 2023-02-14:
Hello @PadmaIoT 

Don't worry; we believed we had addressed all of your concerns, therefore we closed the ticket.

I'll respond to your inquiries in sequence.

1. Please find this line in the file send-class-A-OTAA.ino by going there.

// Set LoRaWAN Class change CLASS_A or CLASS_C
   lora.setDeviceClass(CLASS_A);

It's line 48, Here you can change to class C.

2.- The Beelan LoRaWAN library is designed to work with any Arduino-compatible board that supports the Arduino IDE, so it should be possible to use it with the STM32 Bluepill board as well. While some additional setup may be required, the Beelan LoRaWAN library should be compatible with the STM32 Bluepill board as long as you have the necessary board support files and configurations in place.

3.-If you want to use the Beelan LoRaWAN library in Class C operation with ABP activation, you should use the class-C-ABP.ino file located inside the class-C-ABP folder. This example sketch demonstrates how to configure the Beelan LoRaWAN board for Class C operation with ABP activation and send data to a LoRaWAN network server.

On the other hand, if you want to use the library in a serial communication scenario, you should use the serialToLoRaWAN.ino file located inside the serialToLoRaWAN folder. This example sketch demonstrates how to send data from a serial device, such as a computer, to the Beelan LoRaWAN board and then transmit that data to a LoRaWAN network server.

However, It's important to note that these example sketches are just that - examples. They are meant to demonstrate how to use the Beelan LoRaWAN library in certain scenarios and should be modified to suit your specific needs. You may need to modify the sketches to work with your specific LoRaWAN network server, activation method, or other aspects of your project.

Kind regards!
Support Team.

PadmaIoT, 2023-02-15:
Thanks for your kind support. I used Lorawan Class B OTAA before and LoraWAN is a new technology to me personally. So sometimes ques become so silly. You covered all my ques with answers now. Thank you so much. I appreciate your support.

Eric286, 2023-02-15:
Hello @PadmaIoT 

You're welcome, glad to be of help!

Regarding your question, it is not mandatory to connect the IO pins, RST pin, and NSS/SS pin of the SX12xx LoRa module to Arduino PWM pins specifically. The SX12xx module can be connected to any digital pins of the Arduino board.

However, it is recommended to connect the SX12xx module to certain digital pins of the Arduino board that have specific functionality, as described in the Beelan LoRaWAN library documentation. For example, the library recommends connecting the SX12xx module's NSS/SS pin to Arduino digital pin 10, which is the default SPI slave select (SS) pin on most Arduino boards. This is because the library uses the SPI interface to communicate with the SX12xx module, and connecting the NSS/SS pin to the default SS pin simplifies the SPI communication setup.

Similarly, the library recommends connecting the SX12xx module's RST pin to a dedicated reset pin on the Arduino board, rather than a general-purpose digital pin, to ensure reliable resetting of the module.

In summary, while it is not mandatory to connect the SX12xx module's pins to specific PWM pins on the Arduino board, it is recommended to follow the guidance provided in the Beelan LoRaWAN library documentation to ensure reliable operation of the module with the library.

Best regards!
Support Team.

PadmaIoT, 2023-02-19:
Thanks for your support. Now everything is clear.

Eric286, 2023-02-20:
Hello @PadmaIoT !

You're welcome! I'm glad I could help. If you have any more questions or need further assistance, feel free to ask.

Good luck in your projects.

closed 2023-02-20 by Eric286