ElectronicCats/Beelan-LoRaWAN #151: Problem with US915

AMLP2021, 2023-02-12:
I have been working with EU868 and everything worked ok, but when I changed the frequency to US915 the transmitting node only transmits once. I would like to know if there is a hardware restriction. I am currently using ATMega328 and RFM95

xpeqex, 2023-02-13:
Hi @AMLP2021 

Thank you for contacting us!

In order to understand the issue better we request the following:

- What activation method are you using (ABP, OTAA)?
- The code.

This problem may be caused by a TX Done flag. The "TX Done" flag is a key component of the communication protocol in LoRaWAN that provides feedback to devices on the success or failure of their transmissions, allowing them to take appropriate action to ensure reliable communication with the network server.

I'm looking forward to your reply!

AMLP2021, 2023-02-14:
i'm using ABP.
Hardware: Arduino Nano
Chip Lora: RFM95

Code:

const unsigned long interval = 30000;    // 5 s interval to send message
unsigned long previousMillis = 0;  // will store last time message sent
unsigned int counter = 0;     // message counter

const char* myStr = "" ;
String datos;
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
  .CS = 10,
  .RST = 9,
  .DIO0 = 2,
  .DIO1 = 6,
  .DIO2 = 7,
  .DIO5 = 8,
};

void setup() {
  // Setup loraid access
  Serial.begin(19200);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(12, OUTPUT);
    delay(10000);
  if (!lora.init()) {
    Serial.println("RFM95 not detected");
    while(1);
    delay(5000);
    return;
  }

  // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_A);

  // Set Data Rate
  lora.setDataRate(SF7BW125);

  // set channel to random
  lora.setChannel(1);

  // Put ABP Key and DevAddress here
  lora.setNwkSKey(nwkSKey);
  lora.setAppSKey(appSKey);
  lora.setDevAddr(devAddr);
}

xpeqex, 2023-02-15:
Hi @AMLP2021 

Thank you for the update.

After reading your code I realized that there is nothing wrong with it. That is why we would like you to try the following:

- Try the [Arduino LoRa library](https://github.com/sandeepmistry/arduino-LoRa) made by Sandeep Mistry. Please try the [LoRaSender example](https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples/LoRaSender) configured for the US915 frequency.

The Arduino LoRa library is purely LoRa communication, but this will help us to better identify what is the problem you are having. Please let us know your findings!

xpeqex, 2023-02-20:
Hello @AMLP2021!
This is a message to remind you that your request has been pending for 5 days and awaits your comments. 

Please let us know if your issue is solved or if is there anything else we can assist you with. Otherwise, if we do not hear back from you within the next three days, the ticket will be closed.
:shipit:

wero1414, 2023-02-20:
@AMLP2021 when you configured the config.h for US915 did you activate the sub-band that you want to be using

AMLP2021, 2023-02-26:
> @AMLP2021cuando configuró config.h para US915, activó la sub-banda que desea usar

yes, i activated #define SUBND_0     // 902.3 - 903.7 Mhz

wero1414, 2023-03-01:
So it is probably that your radio is not doing well while transmiting at 915Mhz it could be only for 868Mhz, i suggest you to use the Arduino LoRa library to test just the transmission on that frequency cause we have been using this library for 915 Mhz for a while without a problem.

xpeqex, 2023-03-10:
Hello @AMLP2021 !

We will now close this issue.

However, if you have any doubts or questions, you can reopen the issue or if you think it is necessary, create another one.

Have an amazing day!

Kind regards!
Support Team.

closed 2023-03-10 by xpeqex