ElectronicCats/Beelan-LoRaWAN #153: Soft WDT Reset on lora.sendUpLink()

kcbyrum, 2023-03-03:
I am beelan-LoraWAN version 2.1.0
Board is a Lolin (Wemos) D1 R2 Mini using esp9266 by ESP8266 Community version 3.1.1

The d1 mini is connected to a generic RFM96 transceiver.

lora.init() succeeds
lora.join() succeeds and connects to a gateway on the Senet network

lora.sendUpLink(myStr, strlen(myStr), 0, 1); causes a Soft WDT reset

Any ideas why?

void loop() {
  yield();
  // Check interval overflow
  if(millis() - previousMillis > interval) {
    previousMillis = millis(); 

    sprintf(myStr, "Counter %d", counter); 

    Serial.print("Sending: ");
    Serial.println(myStr);
    
    lora.sendUplink(myStr, strlen(myStr), 0, 1);
    counter++;
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}

Eric286, 2023-03-03:
Hello @kcbyrum 

The Soft WDT reset error is typically caused by the microcontroller's watchdog timer expiring, which happens when the code is stuck in a loop or taking too long to complete a task.

In your code, it looks like you're using the LoRaWAN module to send a message periodically in the loop() function. One possible cause of the Soft WDT reset error could be that the LoRaWAN module is not completing the transmission of the message before the next loop iteration begins. This could lead to an accumulation of unfinished transmissions, causing the code to get stuck in a loop and triggering the watchdog timer.

To avoid this, you can add a delay after the lora.sendUplink() function call to allow the module to finish transmitting the message before proceeding with the next iteration of the loop.

These modifications should help ensure that the LoRaWAN module completes the transmission of the message before the next loop iteration begins, reducing the likelihood of triggering the watchdog timer and causing a Soft WDT reset error.

Best Regards!

Support Team!

kcbyrum, 2023-03-03:
Thanks - I tried putting in a 30 second delay after the uplink and it can send one packet but does the WDT on the second packet.  FWIW, the code runs fine in v1.5.4, it is able to send multiple packets.

kcbyrum, 2023-03-03:
It also resets immediately after sending packet 1, if that helps any

Eric286, 2023-03-07:
Hello @kcbyrum 

If the delay did not help and the Soft WDT reset still occurs, then it's possible that the issue is related to the library or the configuration of the RFM96 transceiver.

Since you mentioned that the code runs fine in v1.5.4, it could be a compatibility issue between the Beelan-LoRaWAN library v2.1.0 and the RFM96 transceiver or ESP8266 board. You could try downgrading to the previous version of the library to see if that resolves the issue.

Another thing you could try is changing the configuration of the RFM96 transceiver, such as adjusting the transmit power or changing the spreading factor. This may help to identify if the issue is related to the configuration of the transceiver.

Try reducing the data rate or changing the frequency to see if that resolves the issue.

Best regards!

kcbyrum, 2023-03-07:
Thanks for the input.  I will experiment with different settings to see if it makes any difference.  It could just be an incompatibility issue.  It is not real reliable about joining to the gateway but i will experiment some more.

Keith

Eric286, 2023-03-14:
Hello @kcbyrum !

Since we have not received a response, we will proceed to close the issue.

In case you have any problems or comments, do not hesitate to contact us again.

Have an amazing day!

Kind regards!

Support Team.

closed 2023-03-14 by Eric286