ElectronicCats/Beelan-LoRaWAN #154: Hello guy, I'm not able to send lora through SF12? Are there any errors here?

tuantran1305, 2023-03-23:
**Please, before reporting any issue**
- Make sure your board it's in good condition.
- Make sure your Module it's in good condition and that it is compatible.
- Verify that it really is a library problem and not a hardware problem.
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any feedback/suggestions should be discussed on the [feedback section](https://github.com/ElectronicCats/Beelan-LoRaWAN/issues):
  * Just click on New Issue and select the correct category.

When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
A clear and concise description of what the bug is.
The more detailed this is, the easier we can come up with a solution.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible.
No extra code, extra hardware, etc.
If you think it is absolutely necessary to add this, mention in detail the connections in case it is necessary hardware and also indicate the part of code that was added and how it was added.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**To help us to understand your situation, we would like to ask you for the following additional information::**
 - Which board are you using?
 - What Operating System and version are you using (e.g. Windows 11, macOS 12.0, Linux)?
 - In case you are using the Arduino IDE, What version of the Arduino IDE?
 - Did it work before?, If it worked before, what were the parameters that you modify?

**Additional context**
Add any other context about the problem here that you think will help us to solve your problem.

Eric286, 2023-03-28:
Dear @tuantran1305 

Thank you very much for contacting us.

With the information you gave me, I will do some tests to verify that everything is correct, in the same way I will report anything to you.

Could you just give me more information please, for example, Are you using The Things Network? Did you move something else in the code? Has it worked for you before?, What exactly happens when you try to send LoRa at SF12? Are you receiving any error messages or notifications?

Have a nice day!

tuantran1305, 2023-04-03:
I use Chirpstack, Gateway, i use the example Class C OTAA, I don't see the join request on the gateway
[image]

tuantran1305, 2023-04-06:
Can anyone help me with this problem?

Eric286, 2023-04-06:
Hello @tuantran1305

I ran some tests with the AS923-2 frequency band and found that only the US, EU, and AUS frequencys seems to be working properly. 

I will be putting this issue in as a bug report and plan to work on a feature to address it in the future.

It's important to note that LoRaWAN networks can vary in their configurations and settings, so it's recommended that you do your own testing and consult with your network operator for guidance on configuring your network for the AS923-2 frequency band.

"Apart from this problem, when I use Class C, Multiple Channel in SF10, AS923-2, I only get downlink confirmed of default frequency 921.4Mhz, and other frequencies do not receive downlink. I checked the Gateway has already sent the downlink, most likely Multi Channel has not been used well in this case"

Based on what you've described, it's possible that the multi-channel settings may not be configured correctly. You may want to double-check the configuration settings on both the node and gateway sides to ensure they are set up correctly for the AS923-2 frequency band.

It's also possible that there may be other factors at play, such as interference or signal strength issues.

Have you tried it with other frequencies?

Best regards!

tuantran1305, 2023-04-06:
I will send the test to you soonest

Eric286, 2023-04-06:
Hello @tuantran1305 

Thank you very much I await your answer.

Best Regards!

Support Team!

tuantran1305, 2023-04-10:
@Eric286 hello bro
I made some tests:
1.SF7-SF10: Node sends data to Gateway--> Chirstack: Done
 SF12: Can't see the data sent in the Gateway log
 2. I tried with all frequencies AS923-2 (921.4->922.8) only 921.4 receive a downlink. But there is a special thing: join request and join accept are both sent and received well at all frequencies in AS923-2

tuantran1305, 2023-04-11:
Can anyone solve this problem?

tuantran1305, 2023-04-17:
Do I need to update any more information? @sabas1080

sabas1080, 2023-04-17:
we are working on it, but we do not have a time frame to resolve this issue.

> Do I need to update any more information? @sabas1080

IoTThinks, 2024-05-25:
I have just checked.
For AS923_2, unable to send uplinks for SF11BW125 and SF12BW125.
I guess it should affect all frequency plans.

I guess something is broken.
Or the library did not handle the AutoLDO when sending SF11 and 12 packets.
Therefore, the gateway doesnot receive the packets. 

I fixed similar issue in another library.
Let me double check for this library too.
https://github.com/SMotlaq/LoRa/pull/11

IoTThinks, 2024-05-26:
This is the pull request.
Yes, need to set Low data rate optimization for SF11/12 for AS923/AS923_2.
I guess should be for all frequency plan.

https://github.com/ElectronicCats/Beelan-LoRaWAN/pull/195

closed 2025-02-19 by sabas1080