ElectronicCats/Beelan-LoRaWAN #155: Stuck on joining network

laura-ebs, 2023-03-30:
Hello!

I am using an Arduino Uno together with a RFM95 SX1276.

I have the following pins connected:
```
const sRFM_pins RFM_pins = {
  .CS = 10,
  .RST = 14,
  .DIO0 = 2,
  .DIO1 = 3,
  .DIO2 = 4,
  .DIO5 = 5,
};
```
This is the full sketch:
```
#include <lorawan.h>

// OTAA credentials
const char *devEui = "xx";
const char *appEui = "xx";
const char *appKey = "xx";

const unsigned long interval = 10000;    // 10 s interval to send message
unsigned long previousMillis = 0;  // will store last time message sent
unsigned int counter = 0;     // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
  .CS = 10,
  .RST = 14,
  .DIO0 = 2,
  .DIO1 = 3,
  .DIO2 = 4,
  .DIO5 = 5,
};

void setup() {
  // Setup loraid access
  Serial.begin(9600);
  while(!Serial);
  if(!lora.init()){
    Serial.println("RFM95 not detected");
    delay(5000);
    return;
  }
  Serial.println("RFM95 detected");

  // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_A);

  // Set Data Rate
  lora.setDataRate(SF9BW125);

  // set channel to random
  lora.setChannel(MULTI);
  
  // Put OTAA Key and DevAddress here
  lora.setDevEUI(devEui);
  lora.setAppEUI(appEui);
  lora.setAppKey(appKey);

  // Join procedure
  bool isJoined;
  do {
    Serial.println("Joining...");
    isJoined = lora.join();
    Serial.println("Joining2");
    //wait for 10s to try again
    delay(10000);
  }while(!isJoined);
  Serial.println("Joined to network");
}

void loop() {
  // Check interval overflow
  if(millis() - previousMillis > interval) {
    previousMillis = millis(); 

    sprintf(myStr, "Counter-%d", counter); 

    Serial.print("Sending: ");
    Serial.println(myStr);
    
    lora.sendUplink(myStr, strlen(myStr), 0, 1);
    counter++;
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}
```
It does detect the chip, it also tries to send the joing request, however it either gets stuck in a loop sending the join requests, or it would only try to send a join request once after which it hangs.

So I modified the 
`lorawan-arduino-rfm.cpp`
to add some debugging info to see where it would get stuck.

The join function looks like this now:
```
bool LoRaWANClass::join(void)
{
    bool join_status;
    const unsigned long timeout = 6000;
    unsigned long prev_millis;

    if (currentChannel == MULTI)
    {
        randomChannel();
    }
    // join request
    Serial.println("Sending join request now");
    LoRa_Send_JoinReq(&OTAA_Data, &LoRa_Settings);
    // delay(900);
    // loop for <timeout> wait for join accept
    prev_millis = millis();
    Serial.println("Waiting for join request now");
    do
    {
        Serial.println("Waiting1");
        join_status = LORA_join_Accept(&Buffer_Rx, &Session_Data, &OTAA_Data, &Message_Rx, &LoRa_Settings);
        Serial.println("Waiting2");

    } while ((millis() - prev_millis) < timeout && !join_status);
    Serial.println("Received or timed out, status: " + String(join_status));
    return join_status;
}
```

And the output inside the console looks like this now:
```
RFM95 detected
Joining...    
Sending join request now
Waiting for join request now
Waiting1
```

Even after like 10 minutes its still stuck there. Also when it doesn't hang I never see the message appear inside TTN console.

Any help would be really appreciated!

xpeqex, 2023-04-04:
Hi @laura-ebs 

Sorry for my late reply!

- Have you tried using the ABP activation?

In OTAA activation, an end device performs a join procedure with a LoRaWAN network, during which a dynamic DevAddr is assigned to an end device, and root keys are utilized to derive session keys. Howeverm in ABP a fixed DevAddr and session keys for a pre-selected network are hardcoded into the end device, and they remain the same throughout the lifetime of an ABP end device. With this mode, an end device skips the join procedure, which seems to be simpler. 

I'm looking forward to your reply!

tuantran1305, 2023-04-11:
@laura-ebs  try with Class C and don't use MULTI, use your default frequency fixed

laura-ebs, 2023-04-11:
Hey, I am sorry for the late reply. I tried to use this module for a school project, however because of the tight deadlines I moved on to another chip. I am sorry for this. If I got some spare time I will try and see if it works with these suggestions but I am not sure when I have some time.

Thanks for the help tho.

closed 2023-04-11 by laura-ebs

xpeqex, 2023-04-11:
@laura-ebs 

No worries at all!

Hope you have success in your school project, and 
do not hesitate to contact us again in case you have any other queries. 💯