ElectronicCats/Beelan-LoRaWAN #156: .setChannel(<int>) doesn't appear to do anything

DeflateAwning, 2023-04-05:
**Describe the bug**
Looking at the code, it doesn't appear that setting an uplink (device-to-network) channel of >7 does anything.

Eric286, 2023-04-06:
Hello @DeflateAwning 

The LoRaWAN protocol defines a set of channels that devices and gateways use to communicate with each other. In general, there are three types of channels:

    Default channels (0-2): reserved for network beacons and are fixed by the LoRaWAN specification.
    Extra channels (3-7): are used for uplink messages from devices to the gateway. These channels are usually configurable by the end device, but must be set in the gateway as well.
    All channels (0-7): include both default and extra channels.

When you call the .setChannel(<int>) function in the code, it should set the current uplink channel to the specified integer value. This means that when the device sends an uplink message, it should use the frequency of the corresponding channel.

In the examples that we have in the repository the line lora.setChannel(MULTI); sets the current uplink channel to a random channel within the allowed channel range.

The MULTI constant is defined in the lorawan.h library and is a value that represents all eight uplink channels. When you pass MULTI as an argument to lora.setChannel(), the LoRaWAN module will randomly select one of the available channels to use for the next uplink transmission.

This is useful in scenarios where you want to avoid using the same channel repeatedly, which could potentially cause interference or collisions with other devices in the network. By randomly selecting a channel for each transmission, you increase the chance of successful transmission without interfering with other devices.

In LoRaWAN, multiple devices share the same channel for communication, so it's important to ensure that the selected channel is not already being used by other devices in the same network. If the selected channel is already occupied, the device won't be able to transmit or receive messages on that channel.

Additionally, some LoRaWAN networks may restrict the use of certain channels, so it's important to check with your network provider or review the network documentation to ensure that the selected channel is allowed for use in the network.

Best Regards!

Support Team.

DeflateAwning, 2023-04-13:
Can you update the documentation to specify the range of valid values of the integer argument? 

Also, how sure are you that 0-7 are the only uplink channels. The Helium Network appears to use other channels in the US915 spectrum for uplink (device-to-network). 

https://docs.helium.com/lorawan-on-helium/frequency-plans/#us915

Eric286, 2023-04-14:
Hello @DeflateAwning 

You can see this in the API section, here it is what it say.

Set Channel

You can set channel allowed in your region (AS_923, EU_868 or US915).

    For US_915 the channels can be [0 - 7]
    Use MULTI if you want random channel

https://github.com/ElectronicCats/Beelan-LoRaWAN/blob/master/API.md

Best Regards!

Support Team.

DeflateAwning, 2023-04-14:
Can you provide a source that confirms that info? The Helium Network appears to use other channels in the US915 spectrum for uplink (device-to-network).

https://docs.helium.com/lorawan-on-helium/frequency-plans/#us915

sabas1080, 2023-04-17:
it is still the same frequency according to TTN, only the channel number has been changed.

https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/

[image]

DeflateAwning, 2023-04-17:
The things network page does not have channel numbers on it. The ordered list numbers aren't channels

Eric286, 2023-04-17:
Dear @DeflateAwning 

The numbers show the relative frequency each channel use.

The numbers provided are not significant in terms of determining which channel to select. Your focus should be on identifying the channel that is used most frequently, regardless of the numerical values associated with each channel's frequency.

You will only need to select the channel's frequency.

Best Regards!

DeflateAwning, 2023-04-17:
The argument to this setChannel function is an integer channel number, not a frequency. Where are the channel number to frequency mappings used for this library?

Eric286, 2023-04-17:
Dear @DeflateAwning 

You can see the void LoRaWANClass::setChannel(unsigned char channel), in the lorawan-arduino-rfm.cpp file in src / arduino-rfm/

In line 338.

Best Regards!

DeflateAwning, 2023-04-17:
Where are the channel number to frequency mappings used for this library? They are obviously not in the [setChannel definition](https://github.com/ElectronicCats/Beelan-LoRaWAN/blob/61bf52455a1c419d1d359218d946d31859787717/src/arduino-rfm/lorawan-arduino-rfm.cpp#L338).

Eric286, 2023-04-18:
Hello @DeflateAwning 

You can see the documentation in https://docs.helium.com/lorawan-on-helium/frequency-plans/ where helium says the channel Number to Frequency (MHZ), and then make a comparison between TTN Frequency to Helium Frequency, https://docs.helium.com/use-the-network/console/migrating-devices/ttn-manual/

In the US915 band, TTN uses a different channel numbering scheme and frequency plan. The TTN channel plan for the US915 band consists of 64 channels, starting from 903.9 MHz up to 927.5 MHz, with a channel spacing of 0.2 MHz. The TTN channel numbers for the US915 band range from 0 to 63.

To convert the frequency for channel 8 in the Helium Network to the corresponding TTN frequency, you can use the following formula:

TTN Channel Number = (Frequency (MHz) - 902.3 MHz) / 0.2 MHz

In this case, the frequency for channel 8 in the US915 band is 903.9 MHz. Therefore, the TTN channel number would be:

TTN Channel Number = (903.9 MHz - 902.3 MHz) / 0.2 MHz = 8

So the TTN channel number for channel 8 in the US915 band is also 8.

The Things Network (TTN) frequency plan page does not provide the channel numbers directly, but instead shows the frequency ranges for the different channels in the US915 band.

To find the channel number for a particular frequency in the US915 band, you can use the following formula:

Channel number = (Frequency (MHz) - 902.3 MHz) / 0.2 MHz

For example, if you want to find the channel number for a frequency of 904.1 MHz in the US915 band, you can calculate:

Channel number = (904.1 MHz - 902.3 MHz) / 0.2 MHz = 9

So the channel number for a frequency of 904.1 MHz in the US915 band is 9.

Have a Nice Day! n.n

DeflateAwning, 2023-04-19:
Source for your conjecture about channel numbering? It should be defined in the LoRa spec regardless of the network, which is where the Helium Network numbering comes from

Eric286, 2023-04-19:
Hello @DeflateAwning 

Our library uses the TTN numbering convention for channels, which is based on the frequency plan used by The Things Network. This is because TTN is one of the most widely used LoRaWAN networks with a large community, and many LoRaWAN devices are already configured to use the TTN frequency plan.

However, if you are using a different LoRaWAN network that uses a different frequency plan, you will need to make changes to the library to reflect the correct frequency plan for your region.

Best Regards!

Support Team.

closed 2023-04-19 by Eric286

DeflateAwning, 2023-04-19:
This issue is not resolved. You have yet to scribe a source for these numberings. Channel numbers are not defined by LoRaWAN networks.

sabas1080, 2023-04-19:
Lora Alliance regional parameters define channels

https://lora-alliance.org/resource_hub/rp2-1-0-3-lorawan-regional-parameters/

[image]

DeflateAwning, 2023-04-20:
This setChannel method in the library sets the uplink frequency; not the downlink frequency. Shouldn't it thus focus on channels 0-63 (BW=125 kHz) and 64-71 (BW=500 kHz)? Why is it currently set to do nothing on valid uplink channel settings?

Eric286, 2023-04-20:
Hello @DeflateAwning 

The library is designed to automatically select an available uplink channel based on the region-specific configuration.

The network server is responsible for setting the frequency for downlink packets, based on the communication parameters negotiated with the end device during the Join procedure.

Kind Regards.