ElectronicCats/Beelan-LoRaWAN #158: unable to join using OTAA

amoat7, 2023-04-22:
Thanks for this library. I joined my rak7289C gateway and my chirpstack server using ABP. My devices are RFM96 and ESP32 firebeetle. However, I'm unable to join using OTAA. Not sure what the issue is. Should I input the devEUI and appKey as MSB or LSB?

Eric286, 2023-04-24:
Hello @amoat7 

"Should I input the devEUI and appKey as MSB or LSB?"

It would have to be most significant bits first, without the 0x format as you can see in the first image.
 
[image]

[image]

Verify that the devEUI, appEUI, and appKey are correct and have been entered in the correct byte order.
Make sure that the frequency plan used by your device and gateway match.
Check the logs of the LoRaWAN server to see if there are any error messages indicating why the device is unable to join.

Kind Regards! 

Support Team!

amoat7, 2023-04-24:
Tried that so many times.. only ABP works for me.

[imagen]<https://user-images.githubusercontent.com/40640735/234115200-3305f730-2bad-48de-9785-a854d49df1eb.png>

[imagen]<https://user-images.githubusercontent.com/40640735/234115214-bc6ece15-4a5d-4fe0-97fd-3de858f4e630.png>

Eric286, 2023-04-24:
Hello @amoat7 

Do you get an error or a message or nothing happens?
What frequency are you using?

The end device must be inside the network's coverage area in order to join an OTAA. This is necessary for the OTAA join operation, which necessitates that the end device be capable of receiving the Join Accept downlink message from the Network Server.

ABP, on the other hand, enables a device to start sending messages as soon as it is turned on, whether or not the device has network coverage.

I tested it with TTN and everything seems to be working correctly.
Could you try it with The Things Network please, to see if the problem persists?

Best Regards!

amoat7, 2023-04-24:
Thanks for the reply but my device is in the coverage area as i can get messages from the esp32 through my network server to my mqtt server using ABP. I’m using my own private chirpstack server, maybe it just isn’t compatible but i don’t think the issue is coverage area. If the frequency (AU915, band 1) didn’t match, ABP won’t also work. I don’t think it’s possible to exchange messages if the node and the server’s frequency and bands are mismatched. There’s no error message, it just says joining.. maybe your library just isn’t compatible yet with chirpstack servers. I’ll just keep using ABP.

Eric286, 2023-04-25:
Dear @amoat7 

We appreciate your feedback and encourage you to stay tuned for updates to our library. We're constantly working to improve compatibility and functionality with a wide range of servers and devices. In our tests, everything appeared to be functioning correctly, so it's possible that there could be other factors at play.

Thanks again for your input, and please don't hesitate to reach out if you have any further questions or concerns.

Best Regards!

closed 2023-04-25 by Eric286