ElectronicCats/Beelan-LoRaWAN #160: Can not send data using OTAA Class-C

PadmaIoT, 2023-05-31:
Hello,
I intend to send and receive data using Class C but except for join request acceptance nothing is happening.
No payload is sent and received. serial monitor says only "joining", nothing else.
 My all configs & details are given Below:

**LoRa Config:**
Network Server: Chirpstack
Frequency Plan: AS923 (edited in .h file to enable desired frequency plan)
Mac Version: LoRaWAN 1.0.2
Revision:  RP002-1.0.3
OTAA: yes
Class: C

**Hardware Config:**
MCU: Arduino Nano
LoRa Chip: RA-01H (SX1276)
(Hardware is ok, also tested by Class B operation using LMIC)

**Code config:**
`lora.setDeviceClass(CLASS_C);
 lora.setDataRate(SF9BW125);
 lora.setChannel(MULTI);`
 
 **Serial monitor:**
 ` Joining...`
 
 **Chirpstack LoraWan frame:**
 
[image]

In this situation What I can do? Can anyone guide me?

AndreaZGuz, 2023-06-02:
Dear @PadmaIoT 

Thank you very much for contacting us.

With the information you gave me, I will do some tests to verify that everything is correct, and in the same way, I will report anything to you.

Could you just give me more information please, for example, did you configure any gateway for your application? Did you move something else in the code? Has it worked for you before?

Have a nice day!

PadmaIoT, 2023-06-03:
Thanks for your kind response. Here is your answers:
1. I am using Tektelic Kona gateway where no separate settings gor class C is needed to config that means I changed nothing.
2. No . I didn't move anything. Not even the payload
3. This is the 1st time I am implementing class c. I never used it before.

Waiting for any breakthrough.
Thanks

AndreaZGuz, 2023-06-09:
Hi @PadmaIoT 

We are now trying to replicate your issue in order to give you the best advice for you to make it work!
As soon as we have any news, we will be informing you.

Thank you in advance for your patience.
Have a nice weekend!

PadmaIoT, 2023-06-15:
Thanks, I am waiting for your update.

AndreaZGuz, 2023-06-15:
Hi @PadmaIoT 

My apologies for the delay in my response.

I ran some tests with the US915_0, EU868, AS923_1, and AS923_2 frequency bands and found all of them except AS923_1 (AS923) are working properly for me on ChirpStack, I just got the same performance as you. I would like to empahtize that in the dashboard, all the time the device were shown as "Active", even when the board were still "Joining" according to my Serial Monitor. 
I will attach a ZIP file containing screenshots of the configurations I did for every one of them, including the status of the packages sent/received, so you can see my outcomes ([OTAA Evidences.zip](https://github.com/ElectronicCats/Beelan-LoRaWAN/files/11761920/OTAA.Evidences.zip)).

You told me before that Class B worked fine. I would like to ask if there is any possibility you can try any of the tests I ran, in order to confirm maybe an issue specifically with AS923 band. Would you be able to do it?

Thanks in advance for your patience.
We are keeping in touch!

PadmaIoT, 2023-06-18:
Thank you so much for the help. I was thinking Should I give you debugging data so that you can find out any issues regarding data reception? And I am pretty sure that if my device can understand the join acceptance then it should work as it should be. For
  class B I used LMIC library. I will run tests and let you know.

AndreaZGuz, 2023-06-21:
Hello again  @PadmaIoT 

If you use our library, RX1 and RX2 windows are fixed to 1 sec. 

Of course! Thanks for trying. We will be waiting for your outcomes.

We are keeping in touch!

PadmaIoT, 2023-06-22:
Hello @AndreaZGuz ,
Will you please let me know that on the time of building this library which version of LoRaWAN documentation is followed? As an example latest documentation of Lora-alliance is for LoRaWAN (not sure) I believe is 1.0.4 and the regional parameter version is RP-002 1.0.3. Let me know, please. I believe something in the library is making this problem. I hope you guys will also try to fix the issue from your side. Happy sharing.

AndreaZGuz, 2023-06-23:
Hello @PadmaIoT,

This library is based on version 1.0.3, it has been tested with version 1.0.4 and worked well, but some features may not be compatible or supported. I have already reported the issue. 
Our team will be working on it to improve the performance of the library.
In the meantime, if you share your results with us once you have run the tests, it would be very helpful to us to identify quickly if the issue is only related to AS923 frequency.

Thank you for your patience and collaboration.

PadmaIoT, 2023-06-25:
Hello @AndreaZGuz ,

Thanks for your continuous support. I have lack of gateways of different bands. I will run the test with our developer community. I already found a gateway of EU868 from one of my developer friend. Will let you know about the result. My request is not close the issue unless it is sloved

AndreaZGuz, 2023-06-27:
Hi @PadmaIoT 

Sure!
We are attentive to your outcomes and comments.

PadmaIoT, 2023-07-02:
Hi, 
I could use only 2 frequency plans. It works fine with EU868. AS923 default (AS923-1) does not work...shows the same problem. The problem is my region only supports AS923.

AndreaZGuz, 2023-07-03:
Hi @PadmaIoT !

Thank you very much for trying.
It seems that we both faced issues with the same frequency. This issue is already reported to our colleagues. Now they will work to solve it.
I cannot assure a specific frame of time to resolve this issue. As soon as we have any updates, we will let you know.
Thanks in advance for your patience and collaboration with the tests.

PadmaIoT, 2023-07-09:
Thanks, @AndreaZGuz . I will be waiting for your solution.

PadmaIoT, 2023-09-07:
Hello there, 
How are you all?
Did you find the bug?

wero1414, 2023-09-07:
Hello we updated with a pull request the way that OTAA class works, can you try again?

AndreaZGuz, 2023-09-28:
Hello @PadmaIoT 
Were you able to test the updates?
Did you face any issues?

TheSnowFall, 2023-09-28:
Hello @AndreaZGuz ,

Thanks for your update. I appreciate your help. Right now I am far away from my lab. I hope very soon I will be able to run and test and let you know about the update.

Good work guys.

PadmaIoT, 2023-09-28:
Hello @AndreaZGuz 

Thanks for your update. I appreciate your help. Right now I am far away from my lab. I hope very soon I will be able to run and test and let you know about the update.

Good work guys.

AndreaZGuz, 2023-09-28:
Hi @PadmaIoT 
Thank you for informing this.
We are attentive to your findings.

halukmy, 2023-10-15:
i can not recieve too anyone fix ?

PadmaIoT, 2024-01-12:
Hi @AndreaZGuz ;
I did not tested it directly, but someone else (one of my fellow developer friends) reported me the problem still persistent. Can you 0lease check again?

AndreaZGuz, 2024-01-12:
Hello @PadmaIoT!

Many thanks for the updates.
Currently, our dev team is working on some updates that will improve the functions from the library, including those related to the AS923 frequency.
The time that these modifications will take is not accurate to tell at this moment, but as soon as they are completed, we will let you know.

Thank you for all the collaboration and your patience.
We are keeping in touch.

damattl, 2024-03-13:
Hi, I had a similar problem.
The uplink was working fine so far, but downlinks where not handled at all.
After trying a few other options I decided to take a look into the implementation and found a flag (_CLASS_C_)
that was disabled / not defined. After enabling it the downlink worked as expected.
I took the freedom to add a comment about this in the example file for OTAA + Class C and started a pull request.

PadmaIoT, 2024-03-13:
@damattl so you mean you solved it somehow?

damattl, 2024-03-13:
@PadmaIoT 
Yes, at least for my project and configuration.
I found this inside the implementation files, and decided to define the `_CLASS_C_` flag via the build flags in the platformio.ini file. 
[image]
[image]
As I'm not using Arduino IDE I don't know how to set build flags in Arduino IDE

PadmaIoT, 2024-03-13:
What you are using? Platform io? @damattl

damattl, 2024-03-15:
Yes, I'm using PlatformIO @PadmaIoT

PadmaIoT, 2024-03-16:
@damattl  so you can do uplink and downlink by class C right? Please give me more insight about your configuration of the C class device in  LoRa server. I want to know which LoRA server you are using, and when you register your device in the server, what was the configuration you applied. I will be thankful.

ryandiapsl, 2024-03-22:
I also have an issue, i attempted to use this library and have imported the new, *fixed library. I setup the end node with the example of Class C OTAA on platform IO and have set all the build flags. I have managed to config my end device on 2 different servers that being TTN and Datacake and managed to make a join connection from my end device to my gateway and also send uplinks and receive them on TTN and Datacake, but for some reason downlinks never work on both of them. I am using ttgo-lora32-v1 with SX1276 lora module.

AndreaZGuz, 2024-03-22:
Hello!

Thank you for following up on this issue and helping us with your feedback and comments.
At the moment, some changes have been implemented to enhance the efficiency of this library. The last update we can point you to is [pull-request 181](https://github.com/ElectronicCats/Beelan-LoRaWAN/pull/181).
Due to other priorities, our team will resume work in the next half of the year.

We appreciate your patience and any collaboration. Remember, if you want to contribute to this library, you can always open a pull request with your changes/suggestions.

aarnaud, 2024-04-17:
first, thanks for your work.

I'm using LoRa32 V2.1_1.6, I success to use OTAA after connecting lora1 (DIO1) to the esp32. I also connected lora2 (DIO2).

I can joined Helium network (US-915) and send data but It failed to receive any data from the downlink.

The received is working correctly in a p2p setup.

I tested the latest release, master and the branch linked to the PR 181.

Is it something link to RX1 or RX2 frequency for US-915 ?

AndreaZGuz, 2024-04-17:
Hello @aarnaud,

It may be related to the Class you are using. 
Are you using Class A or Class C?

aarnaud, 2024-04-17:
Hi @AndreaZGuz, sorry for this missing info, Class C, I want to receive data from helium(using Meteo Scientific), I also have a hotspot, so both are not fare away.

AndreaZGuz, 2024-04-17:
Hi @aarnaud !

The improvements for this library include some features for Class C.
This behavior is expected. Our team will continue working to fix these issues in the next half of the year.

We apologize for the inconvenience and appreciate any contribution.

IoTThinks, 2024-05-26:
This should fix the Joining issue for AS923.
https://github.com/ElectronicCats/Beelan-LoRaWAN/pull/192

Khaiminh2310, 2025-02-13:
Hi @PadmaIoT @damattl, I'm working on this library for my project (AS923-2, Class C). I am quite familiar with this library and have also modified quite a bit to make uplink and downlink work well with ChirpStack server. In this topic if you are still having problems with uplink or downlink in class C, you can leave detailed information about the problem and I will be happy to support.

aarnaud, 2025-02-13:
> Hi [@PadmaIoT](https://github.com/PadmaIoT) [@damattl](https://github.com/damattl), I'm working on this library for my project (AS923-2, Class C). I am quite familiar with this library and have also [...]

> Receiving downlink is still an issue. Will check if can fix it.
This feature work before but recent update brokes the downlink.

so there is no issue anymore with this  ?

PadmaIoT, 2025-02-13:
> Hi @PadmaIoT @damattl, I'm working on this library for my project (AS923-2, Class C). I am quite familiar with this library and have also modified quite a bit to make uplink and downlink work well [...]

Hi @Khaiminh2310 ,
Thwt is excellent. Can you enlightened us how your fix is working?

Khaiminh2310, 2025-02-13:
> so there is no issue anymore with this ?

There is currently no release that guarantees no issues for uplink and downlink for this library. I have some internal modifications based on the v2.5.0 tag to make my project work well both uplink and downlink. If you have any specific issues, I can double check and give you some advise.