ElectronicCats/Beelan-LoRaWAN #161: Getting ACK message for uplink message

kadirozdinc, 2023-06-07:
Hello, I can make ClassC/OTAA with Chirpstack Server thanks for your effort, I have a question about confirmed messages. I try to get ACK message back for every uplink I wanna make sure uplinks arrived successfully at my node side. Codes I try are below : 

```
#include <Arduino.h>
#include <lorawan.h>

// OTAA credentials
const char *devEui = "80B3D57ED004397D";
const char *appEui = "1300000000000013";
const char *appKey = "47521E11573093C237C7333983DD475C";

const unsigned long interval = 15000; // 10 s interval to send message
unsigned long previousMillis = 0;     // will store last time message sent
unsigned int counter = 0;             // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

#define LED 1

const sRFM_pins RFM_pins = {
    .CS = 7,
    .RST = 3,
    .DIO0 = 18,
    .DIO1 = 19,
    // .DIO2 = -1,
    // .DIO5 = -1,
};

void setup()
{
  // Setup loraid access
  Serial.begin(115200);
  while (!Serial)
    ;
  if (!lora.init())
  {
    Serial.println("RFM95 not detected");
    delay(5000);
    return;
  }
  pinMode(LED, OUTPUT);
  // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_C);

  // Set Data Rate
  lora.setDataRate(SF8BW125);

  // set channel to random
  lora.setChannel(MULTI);

  // Put OTAA Key and DevAddress here
  lora.setDevEUI(devEui);
  lora.setAppEUI(appEui);
  lora.setAppKey(appKey);

  // Join procedure
  bool isJoined;
  do
  {
    Serial.println("Joining...");
    isJoined = lora.join();

    // wait for 10s to try again
    delay(10000);
  } while (!isJoined);
  Serial.println("Joined to network");

  // sprintf(myStr, "First-%d", counter);

  // Serial.print("Sending: ");
  // Serial.println(myStr);
  // lora.sendUplink(myStr, strlen(myStr), 1, 1);
}

void loop()
{
  // Check interval overflow
  if (millis() - previousMillis > interval)
  {
    previousMillis = millis();

    sprintf(myStr, "Counter-%d", counter);

    Serial.print("Sending: ");
    Serial.println(myStr);
    //data,length,confirmed?,port
    lora.sendUplink(myStr, strlen(myStr), 1, 1);
    counter++;
     Serial.println(lora.readAck());
  }
  
  if(lora.readAck()) Serial.println("ackk");

  recvStatus = lora.readData(outStr);
  if (recvStatus)
  {
    Serial.println(lora.getRssi());
    Serial.println(outStr);
    if (outStr[0] == 0x41) // A gelmişse led on
      digitalWrite(LED, HIGH);
    else
      digitalWrite(LED, LOW); // farklı veri ise off
  }

  // Check Lora RX
  lora.update();
  

}
```

Where do I make a mistake ? How I can make it ? Thanks in advance

AndreaZGuz, 2023-06-07:
Hello, @kadirozdinc !

Many thanks for contacting us and for the information.

I have tested your code and it successfully worked for me except for the output that turns on the LED.
If I send a downlink message from the server with 0x41, I am able to read "A" in the Serial Monitor, but the led never turned on. I suggest you add to the void setup() the first state for the output you want to turn on when you receive "A" from the server. It means, after pinMode(LED, OUTPUT); please add the instruction digitalWrite(LED, LOW);

I would like to clarify that I used TTN as my server to test the application. Please, try it with your server and let me know if you finally solved your issue or if there is anything else I can help you with.

We are keeping in touch.

kadirozdinc, 2023-06-08:
Thanks for your interest .Codes works for me successfully I can turn the led on no problem but what I want to emphasie was different issue. I try to get ACK messages for every uplink messages. You know there are 2 types of payload which are confirmed and unconfirmed. When I set it as confirmed, at node side I need to get ACK message which means uplink arrived network server without problem. Briefly I want to see ACK messages in arduino serial.

I looked over library I found a function named readAck() but I cannot make it work.

AndreaZGuz, 2023-06-08:
Hi @kadirozdinc !
Thanks for clarifying, now I understand.
In order to try to replicate your issue, could you please share with me the following additional information:

1. Are you still using Class C as your code is written?
2. Did you confirm that downlink messages are being sent every time the server receives data from your node?
3. Are the uplink messages stable and continuously reaching the server?

kadirozdinc, 2023-06-12:
1- Yes I use in Class C
2- No, Whenever I send downlink from Chirpstack, I can get it in node side without any problem, in class C there is no obligation to get downlink messages when send an uplink.
3- Yes I can observe uplink message in network server.

kadirozdinc, 2023-06-20:
Any idea ?

AndreaZGuz, 2023-06-20:
Hi again @kadirozdinc !

We are reviewing the issue based on your last info in order to see where the issue could be originated.
As soon as we have information related to it, we will be writing back to you.

Thank you for your patience, collaboration, and information.
We are keeping in touch.

AndreaZGuz, 2023-06-28:
Hi @kadirozdinc !

I have already run some tests for this issue.

Using OTAA, I was never able to receive any ACK message, after executing the uplink. I started trying with the same specifications you are using (Class C, I used US_915 FSB 2 for TTN and FSB 1 for ChirpStack). No success.

I tried using ABP mode.
1.- TTN: I registered a device to work with ABP mode, US_915, FSB 2, Class A. It worked.
2.- ChirpStack: I registered a device to work with ABP mode, US_915, FSB 1, Class A. It did not work.

I will attach a ZIP file where I added some screenshots.
[ACK issue.zip](https://github.com/ElectronicCats/Beelan-LoRaWAN/files/11900122/ACK.issue.zip)

I wonder if you are able to execute a test maybe with TTN using the same specifications I did use. This way we can confirm that maybe the issue is with OTAA mode and, in addition, there is something else to do with ChirpStarck in order to get ACK using the same settings as TTN.
This library is based on version 1.0.3, it has been tested with version 1.0.4 and worked well, but some features may not be compatible or supported. I have already reported the issue. Now, I cannot assure a specific frame of time to resolve this issue. As soon as we have any updates, we will let you know.

We are keeping in touch!

AndreaZGuz, 2023-07-10:
Hello @kadirozdinc 

If you do not have more questions or findings at this moment, we will now close this issue.
Remember you can re-open it if you need further assistance.

Regards!

sabas1080, 2023-07-21:
#118

kadirozdinc, 2023-07-24:
I apply the change #118 but it didn't work for me somehow. It would be perfect if there is an example about how read ack after sending uplink with classA and classC

kadirozdinc, 2023-07-24:
I have made some test and there are my inferences :

- Firstly in Class C Mode, it seems not possible to readAck() since **Ack_Status = NEW_ACK**; line works only in Class A mode.
- Network Server sends downlink including ACK message in RX1 window. In the Class C mode, right after sending uplink, RX2 window opens and remain so till the next transmission. (which means we will catch that message) Normally according to lorawan specifications, Class C mode should include Class A mode behaviour. 
Refer to : [](https://lora-developers.semtech.com/documentation/tech-papers-and-guides/lorawan-class-c-devices/)
- When I switched to Class A, after some modification, I could get ack downlink message in RX1 windows and get real data in RX2 windows. After turning the gateway off, I could not get any data ( which means success)

I think it is necessary to make an arrangement according to Lorawan Spec. There are my observation, if I am wrong, please correct me. If I made a progress, I will inform you and pull request

Thanks for your attention.

AndreaZGuz, 2023-07-25:
Hi @kadirozdinc !

Thank you for your feedback! 

We are still running some tests to get as much information as possible to find the best solution for this issue. 
The last test we executed threw the results explained before.

However, we will be grateful for any progress made on your own and shared with us by making a pull request. 
We will be informing you about any updates on the library.

We appreciate your collaboration.