ElectronicCats/Beelan-LoRaWAN #166: Join procedure EU

sky59, 2023-07-10:
Hello,

I am not 100% sure but according to the lorawan specification page 35 table 13 those are the 6 frequencies to join wan.

But in code there are defined MULTI channels that means any out of 

    { 0xD9, 0x06, 0x8B }, //Channel [0], 868.1 MHz / 61.035 Hz = 14222987 = 0xD9068B
    { 0xD9, 0x13, 0x58 }, //Channel [1], 868.3 MHz / 61.035 Hz = 14226264 = 0xD91358
    { 0xD9, 0x20, 0x24 }, //Channel [2], 868.5 MHz / 61.035 Hz = 14229540 = 0xD92024
    { 0xD8, 0xC6, 0x8B }, //Channel [3], 867.1 MHz / 61.035 Hz = 14206603 = 0xD8C68B
    { 0xD8, 0xD3, 0x58 }, //Channel [4], 867.3 MHz / 61.035 Hz = 14209880 = 0xD8D358
    { 0xD8, 0xE0, 0x24 }, //Channel [5], 867.5 MHz / 61.035 Hz = 14213156 = 0xD8E024
    { 0xD8, 0xEC, 0xF1 }, //Channel [6], 867.7 MHz / 61.035 Hz = 14216433 = 0xD8ECF1
    { 0xD8, 0xF9, 0xBE }, //Channel [7], 867.9 MHz / 61.035 Hz = 14219710 = 0xD8F9BE

so these lines:

#elif defined(EU_868)    
    freq_idx = random(0,8);
    LoRa_Settings.Channel_Rx=freq_idx;      // same rx and tx channel 

can return also frequensies 867 for join but it is not correct? Only first 3 channels are allowed

xpeqex, 2023-07-10:
Hello @sky59 

Thank you for reaching out to us.

We appreciate your bringing this to our attention, we would like to understand better this issue/discrepancy, so please share with us the link of the LoRaWAN specification document with which you obtained the table you mentioned. 

We request this information since we are not sure what version of the LoRaWAN Specification documentation are you referring to.

Greetings.

sky59, 2023-07-18:
[image]

xpeqex, 2023-07-19:
@sky59 
Thank you for sharing this screenshot with us.

I will refer my reply to the [LoRaWAN Regional Parameters](https://resources.lora-alliance.org/home/rp002-1-0-4-regional-parameters) documentation version RP002-1.0.4.

Section **2.4.2 EU863-870 Band Channel Frequencies**.

> The following table gives the list of frequencies that SHALL be used by end-devices to broadcast the Join-Request message. The Join-Request message transmit duty-cycle SHALL follow the rules described in chapter “Retransmissions back-off” of the LoRaWAN [TS001] specification document.
> [image]

Section **2.4.4 EU863-870 Join-Accept CFList** (Channel Frequency List)

> The EU863-870 band LoRaWAN implements an OPTIONAL CFList of 16 octets in the Join Accept message for both CFListType 0 and CFListType1. When receiving a CFListType 1, it SHALL be interpreted according to the channel list definition in Section 2.3.2.1 (800 MHz Fixed Channel List Definition)

Section **2.3.2.1 800 MHz Fixed Channel List Definition**

> The 800 MHz fixed channel list defines 40 channels that may be created. Channel-ID 0 is 863.1 MHz; channels increase at 200 kHz spacing to Channel-ID 34 at 869.9 MHz. Five additional channels, identified as 35 to 39, are defined at 865.0625 MHz, 865.4025 MHz, 865.6025 MHz, 865.785 MHz.

sky59, 2023-07-20:
Thank you for dealing with this "problem"

But am I right at this moment in the code is defined MULTI even for join procedure that means any of 8 defined channels can be used that is not correct? (see my original input) Should not it be only first three as you posted screenshot?

xpeqex, 2023-07-20:
@sky59 

The library uses the definition MULTI (channels) for all the functions and not exclusively for the JOIN. It is true that the Join process is must be only over the three defined frequencies by the LoRaWAN Alliance. 

If we take a look at the library code in the `lorawan-arduino-rfm.cpp` file, the class join uses MULTI to set the join request channel, which includes all the defined frequencies channels in the library and the frequencies 868.10, 868.30 and 868.50 MHz. Meaning that at some point of the join process the library will use any of these three frequencies to send the join request.

We invite you to share your findings or modifications that enhance the functionality of the library with these particular channels frequencies opening a Pull request.

Greetings

sky59, 2023-07-21:
Sorry, I am lost. Is the code correct or not? (I think not correct)

Anyway, I modified library for 433MHz usage "the easy way" 👍  Using only first 3 frequencies, though I defined all
---------------------------------------------------------------------------------------------------------------
void LoRaWANClass::randomChannel()
{
    unsigned char freq_idx;
#ifdef AS_923
    freq_idx = random(0, 9);
    // limit drate, ch 8 -> sf7bw250
    LoRa_Settings.Datarate_Tx = freq_idx == 0x08? 0x06 : drate_common;
#elif defined(AS_923_2)
    freq_idx = random(0, 8);
	LoRa_Settings.Channel_Rx=freq_idx;      // same rx and tx channel
#elif defined(EU_433)                       // ** sky59 **
    freq_idx = random(0,3);                 // bolo (0,8) ale iba 3 prve kanaly pouzijem koli Join
    LoRa_Settings.Channel_Rx=freq_idx;      // same rx and tx channel 
#else // US_915 or AU_915
    freq_idx = random(0, 8);
    LoRa_Settings.Channel_Rx = freq_idx + 0x08;
#endif
    LoRa_Settings.Channel_Tx = freq_idx;
}

-----------------------------------------------------------------------------------------------------------------------
  static const PROGMEM unsigned char LoRa_Frequency[9][3] = {//[433.175 - 434.665] MHz
//  32MHz / 2exp(19) = 61.03515625 Hz  ** sky59 **
    { 0x6c, 0x4b, 0x33 }, //Channel [0], 433.175 MHz / 61.03515625 Hz = 0x6C4B33   pouzijem iba toto aj pre Join
    { 0x6c, 0x58, 0x00 }, //Channel [1], 433.375 MHz / 61.03515625 Hz = 0x6C5800   pouzijem iba toto aj pre Join
    { 0x6c, 0x64, 0xcc }, //Channel [2], 433.575 MHz / 61.03515625 Hz = 0x6C64CC   pouzijem iba toto aj pre Join
    { 0x6c, 0x71, 0x99 }, //Channel [3], 433.775 MHz / 61.03515625 Hz = 0x6C7199   nepouzit
    { 0x6c, 0x7e, 0x66 }, //Channel [4], 433.975 MHz / 61.03515625 Hz = 0x6C7E66   nepouzit
    { 0x6c, 0x8b, 0x33 }, //Channel [5], 434.175 MHz / 61.03515625 Hz = 0x6C8B33   nepouzit
    { 0x6c, 0x98, 0x00 }, //Channel [6], 434.375 MHz / 61.03515625 Hz = 0x6C9800   nepouzit
    { 0x6c, 0xa4, 0xcc }, //Channel [7], 434.575 MHz / 61.03515625 Hz = 0x6CA4CC   nepouzit
    { 0x6c, 0xaa, 0x8f }, // RX2 Receive channel 434.665 MHz / 61.03515625 Hz = 0x6CAA8F   pevna frekvencia
  };

sky59, 2023-07-21:
I am still waiting for some components so I could not test 433 yet. I am in Klosterneuburg and on the thethingsnetwork.org/map it shows a gateway should be present, but I have no idea if it is also for 433MHz - I can not find this info anywhere. There are no buildings so I took ma bicycle and went there. What I found was surprise also for me, so I give it a big chance! :) See photos
[image]
[image]

xpeqex, 2023-07-21:
@sky59 

The code is correct, but not perfect, since the library supports various bands and channels it is way easier to use MULTI instead of writing each frequency channels for each supported bands.

Additionally, we really appreciate you shared your modification, if you have any other suggestions please let us know.

sky59, 2023-07-22:
Really sorry being "ignorant" but if a random function returns number 3,4,5,6,7 (because of MULTI) then selected frequency is not correct for Join procedure, or?
Then it must be repeated another Join attempt until the random function returns 0,1 or 2?

xpeqex, 2023-07-24:
@sky59 Correct, the function will be repeated until the correct frequency channel is selected, and it performs the join process. We really understand that in order to join the channels are already defined by LoRa alliance for each band.

Our library, which you can modify to fit your needs, uses MULTI which choose a random frequency of the band to perform the LoRaWAN functions, not only the Join process, making it not necessary to be testing the frequencies one by one to know which one works correctly. Hope you can understand we are doing this to provide a more user-friendly user of the library.

sky59, 2023-07-25:
Thank you for explanation.

I have a last question as you know much more than me : where can I find info about public gateways? thethingsinternet/map does not provide any good information

xpeqex, 2023-07-25:
@sky59 
This information depends entirely on the LoRaWAN Network service you use (TTN, Chirpstack, etc), so we cannot give an accurate or approximate information. There is a website called ttnmapper, you may find it helpful but as its name states this is a map only for the registered gateways in TTN and may not be up-to-date.

Lastly, we appreciate your observations you made on the library and your feedback. Let us know if you have any other questions.

xpeqex, 2023-07-28:
@sky59 

We will close this issue now, however, please feel free to reach out to us again in case you have further suggestions, encounter an issue, or you would like to contribute.

We eagerly anticipate witnessing the remarkable creations you will bring forth! Visit [hackster.io/ElectronicCats](http://www.hackster.io/ElectronicCats), where you can see others projects and share yours!

Wishing you a fantastic day ahead!

closed 2023-07-28 by xpeqex