ElectronicCats/Beelan-LoRaWAN #167: Issue setting up correct pins for ESP8266

GRpro, 2023-07-12:
Hello, 
Trying to use the library for RFM95_V4.0 and NodeMCU ESP8266 and having issues.

My pin config might be wrong, but this is what I use
```
const sRFM_pins RFM_pins = {
  .CS = D5,
  .RST = D6,
  .DIO0 = D0,
  .DIO1 = D1,
  .DIO2 = D2,
  .DIO5 = D3,
};
```

OOM issue
```
RFM95 not detected
Sending: Counter-0

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4010030c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc70 end: 3fffffd0 offset: 0160
...
<<<stack<<<

```

Could you please help providing correct pin config for RFM95 module, thanks!

Eric286, 2023-07-12:
Hello @GRpro !

Thank you for contacting us!

Can you please try this pin configuration?

const sRFM_pins RFM_pins = { .CS = 20, .RST = 9, .DIO0 = 0, .DIO1 = 1, .DIO2 = 2, .DIO5 = 15, };

Also, please tell us What activation method are you using (ABP, OTAA)?, and sen us your code.
Can you send us a schematic of your connection?

Best regards!

GRpro, 2023-07-13:
Hey @Eric286, thanks for reaching back!

In your connection I don't see that SPI interface is connected fully, is that expected?
I'm using OTAA method

Here is the code

```
/**
 * Example of OTAA device      
 * Authors: 
 *        Ivan Moreno
 *        Eduardo Contreras
 *  June 2019
 * 
 * This code is beerware; if you see me (or any other collaborator 
 * member) at the local, and you've found our code helpful, 
 * please buy us a round!
 * Distributed as-is; no warranty is given.
 */
#include <lorawan.h>

// OTAA credentials
const char *devEui = "...";
const char *appEui = "...";
const char *appKey = "...";

const unsigned long interval = 10000;    // 10 s interval to send message
unsigned long previousMillis = 0;  // will store last time message sent
unsigned int counter = 0;     // message counter

char myStr[50];
char outStr[255];
byte recvStatus = 0;

const sRFM_pins RFM_pins = {
//  .CS = 20,
//  .RST = 9,
//  .DIO0 = 0,
//  .DIO1 = 1,
//  .DIO2 = 2,
//  .DIO5 = 15,
  
//  .CS = D5,
//  .RST = D6,
//  .DIO0 = D0,
//  .DIO1 = D1,
//  .DIO2 = D2,
//  .DIO5 = D3,

  .CS = 20, 
  .RST = 9, 
  .DIO0 = 0, 
  .DIO1 = 1, 
  .DIO2 = 2, 
  .DIO5 = 15,
};

void setup() {
  // Setup loraid access
  Serial.begin(9600);
  while(!Serial);
  if(!lora.init()){
    Serial.println("RFM95 not detected");
    delay(5000);
    return;
  }

  // Set LoRaWAN Class change CLASS_A or CLASS_C
  lora.setDeviceClass(CLASS_A);

  // Set Data Rate
  lora.setDataRate(SF9BW125);

  // set channel to random
  lora.setChannel(MULTI);
  
  // Put OTAA Key and DevAddress here
  lora.setDevEUI(devEui);
  lora.setAppEUI(appEui);
  lora.setAppKey(appKey);

  // Join procedure
  bool isJoined;
  do {
    Serial.println("Joining...");
    isJoined = lora.join();
    
    //wait for 10s to try again
    delay(10000);
  }while(!isJoined);
  Serial.println("Joined to network");
}

void loop() {
  // Check interval overflow
  if(millis() - previousMillis > interval) {
    previousMillis = millis(); 

    sprintf(myStr, "Counter-%d", counter); 

    Serial.print("Sending: ");
    Serial.println(myStr);
    
    lora.sendUplink(myStr, strlen(myStr), 0, 1);
    counter++;
  }

  recvStatus = lora.readData(outStr);
  if(recvStatus) {
    Serial.println(outStr);
  }
  
  // Check Lora RX
  lora.update();
}
```

Can you please point out which connections should I make?
Not sure what is pin 20 for ESP8266
```
  .CS = 20, // ???
  .RST = 9, //  SD2 - GPIO 9 connect to RESET on module 
  .DIO0 = 0, // D3 - GPIO0
  .DIO1 = 1, // TX - GPIO1
  .DIO2 = 2, // D4 - GPIO2
  .DIO5 = 15, // D8 - GPIO15
```

Attaching photos of used hardware
[image]
[image]
[image]

Eric286, 2023-07-13:
Hi @GRpro!

_**"In your connection I don't see that SPI interface is connected fully, is that expected?"**_

There is no pin standard; we have utilized the pins found on our development boards, however changing the pins on your devboard is a straightforward process.

_**Can you please point out which connections should I make?**_

You can change that pin if you want.

There were some previous tests with ESP8266 and a RFM96
I will try to do some tests with the RFM95 to be sure everything works ok.

do you have anther board where you can test the module to see if it is really working?

Here's a common pin mapping for the NodeMCU ESP8266 board with the RFM95 module:
```

const sRFM_pins RFM_pins = {
  .CS = D8,
  .RST = D4,
  .DIO0 = D1,
  .DIO1 = D2,
  .DIO2 = D3,
  .DIO5 = D5,
};
```

Double-check your connections and try running the code again to see if the issue persists.

Kind Regards!

Support Team

GRpro, 2023-07-13:
Thanks for the answer.
It looks like my module requires 5V power supply VCC, so I use the shield for ESP8266 to make it work. 
[RFM95PW.pdf](https://github.com/ElectronicCats/Beelan-LoRaWAN/files/12043980/RFM95PW.pdf)
`.CS` should be connected to `NSS` ?

I have Arduino UNO and ESP32 to test if it works.
Could you please provide pin config for both boards above?

GRpro, 2023-07-14:
@Eric286 , it doesn't work.
The only output I get is 
```
rl l⸮r$⸮n⸮l⸮b|⸮⸮⸮rb⸮b⸮nnlnn2bbpp⸮$blrl
```

The print section at the beginning of setup function (which I've added) prints nothing

Eric286, 2023-07-14:
Hello @GRpro !

First we are going to peform a test in the Arduino UNO to see if the module it is working.

Use this pin configuration:

```
const sRFM_pins RFM_pins = {
    .CS = 10,
    .RST = 9,
    .DIO0 = 2,
    .DIO1 = 3,
    .DIO2 = 4,
    .DIO5 = -1,
}; 
```

It has already been tested and works correctly.

Also please make sure that the necessary voltage is reaching the module.

Kind Regards!

Support Team

GRpro, 2023-07-15:
@Eric286 , I've connected 
For this pin configuration
```
const sRFM_pins RFM_pins = {
    .CS = 10,
    .RST = 9,
    .DIO0 = 2,
    .DIO1 = 3,
    .DIO2 = 4,
    .DIO5 = -1,
}; 
```
I use following wiring
```
All GND -> GND
3.3V -> 3.3V
NSS ->  10 (Also tried SCK -> 10)
RESET -> 9
DIO0 -> 2
DIO1 -> 3
DIO2 -> 4
```

Tried with a more widespread RFM95W module
[image]

Have the following issue
```
RFM95 not detected
```

Eric286, 2023-07-18:
Hello @GRpro!

I perform some test with the Arduino UNO and the same module that you have (RFM95W), and everything is working fine, can you please check your connections to see if there is any error. 

Also, could you also send me an image of your board with the connections? to see if I can see the error.

Kind Regards!

Support Team

GRpro, 2023-07-18:
@Eric286 , It would be really helpful if you can post an image of your connection, or better specify them in a format 
(RFM95W PIN -> Arduino PIN) as I did in my last comment. 
I still don't understand if you connected SCK and NSS module outputs, also what about MOSI and MISO

Could you please elaborate? Thanks

Eric286, 2023-07-26:
Hello @GRpro 

Due to the lack of response, we will close the issue.

If you encounter any further difficulties or wish to share any feedback, please feel free to reach out to us again. We are here to assist you.
Wishing you a fantastic day ahead!

Kind regards,

closed 2023-07-26 by Eric286