ElectronicCats/Beelan-LoRaWAN #168: Method to get DevAddr, NwkSKey and AppSKey

LioBou, 2023-07-16:
Hi,
first, thank you for this very nice and useful librairie ! I use it with RFM95 and Raspberry Pico µcontroller, works nice.

Suggestion that would help when the reception (gateway to device) is not always good and JOIN procedure fails often:

in OTAA mode, since the JOIN procedure is not compulsory each time you want to send data, I would like to be able to use the previous valid DevAddr, NwkSKey and AppSKey parameters saved during the last successful JOIN.

For that, a method like these needs to be added to the librarie : 
**lora.getDevAddr();
lora.getNwkSKey();
lora.getAppSKey();**

then it would be possible to re-use those values for the following send, using the existing method:
lora.setDevAddr(devAddr);
lora.setNwkSKey(nwkSKey);
lora.setAppSKey(appSKey);

I believe these 3 values are saved somewhere in the code and new methods need to be developped (unfortunately, I'm not good enough at coding :( to do it)

Would it be possible to add these methods ?
Thank you !

xpeqex, 2023-07-19:
Hello @LioBou 

We appreciate your interest in our library.

Indeed, the library has the ability to obtain this information, however, is not totally exposed in the code. 
We encourage you to explore the capabilities of our library on your own and verify this. If you find something interesting we invite you to post it here and share it with us and the community.

Greeting

xpeqex, 2023-07-28:
Hi @LioBou 

Since we didn't receive an update on your end we will close this ticket.

If you encounter any further difficulties or wish to share any feedback, please feel free to reach out to us again. We are here to assist you.

closed 2023-07-28 by xpeqex

wero1414, 2023-07-28:
Hi @LioBou you can find more information about this on #130 we talked about where the credentiuals can be found