ElectronicCats/Beelan-LoRaWAN #171: Poor ESP8266 support since 2.0.0 and up

virtyaluk, 2023-07-23:
Hello there!

Despite poorly written documentation, I managed to get this lib up and running with my NodeMCU v3 and RFM95W. I had to play around with different wiring setups and utilize different library APIs from different versions to have my end-node device running without errors for the last 1.5 days.

The main drawback of the lib that makes it unsuitable for production is that starting from version 2.0.0 and newer it doesn't support ESP8266-based MCUs. To be clear, it does support ESP8266, but a bug in an API call to `lora.update` results in a constant crash loop. I managed to track the troubled code place back to the `LORA_Cycle` call that leads to a semi-infinite while loop that makes an MCU crash and reset. I've seen a couple of other postings regarding the issue ([here](https://github.com/ElectronicCats/Beelan-LoRaWAN/issues/54), [here](https://github.com/ElectronicCats/Beelan-LoRaWAN/issues/62) and [here](https://github.com/ElectronicCats/Beelan-LoRaWAN/issues/77)) and even a PR to fix that, but it seems like the patched code was reverted at some point. In my case, I had to downgrade back to version 1.5.4 which works like a charm but has a limited set of APIs of particular interest to me (waking up mechanism and the like). Therefore, I conclude that **this lib is not ready for production use**.

What surprised me the most was that the ESP32 setup went big in just a single try. No single error or warning. And for some unknown reason, it worked much faster rather than the ESP8266 setup, meaning it established a connection to a gateway in seconds and wasn't swamping the gateway with tens of JoinRequests.

Thanks.

sabas1080, 2023-07-23:
Hello @virtyaluk 

I understand your problem but apparently in version 2.3.0 we have fixed that problem, but it seems to be the library manager did not detect the change, can you please try from this link and tell us your results, also I will check the update problem. https://github.com/ElectronicCats/Beelan-LoRaWAN/releases/tag/v2.3.0

sabas1080, 2023-07-23:
the problem has been fixed, version 2.4.0 is now available, thanks for your feedback.

closed 2023-07-23 by sabas1080